terraref/reference-data #192: Upload 2016 PhenoTractor data to trait database

dlebauer, 2017-09-21:
@dlebauer commented on [Thu Sep 21 2017](https://github.com/terraref/computing-pipeline/issues/358)

The earlier issue https://github.com/terraref/computing-pipeline/issues/211 was closed, apparently before the data was inserted into the trait database. 

Data are in https://docs.google.com/spreadsheets/d/1NNKtmWFBdoh2iaaXZh9fsu_nNRLMpytJ8Av1bHxpgIg/edit#gid=698446872 Need to be uploaded

Method = 'Maricopa Phenotractor'
Citation title = 'Maricopa Field Station Data and Metadata'

### Completion Criteria

* [ ] reformat data as needed for upload
* [ ] upload data from four dates in the spreadsheet.
* [ ] upload raw datasets to Clowder (?)

---

@NewcombMaria commented on [Thu Sep 21 2017](https://github.com/terraref/computing-pipeline/issues/358#issuecomment-331207595)

I can help with the steps to reformat data and upload the data to betydb.  I'll talk with Pedro and we'll make a plan.  I've never uploaded to clowder but if that's of interest I can learn with some guidance from someone in Illinois.

NewcombMaria, 2017-09-21:
I've reformatted the tractor data from the 4 dates and it won't take me long to upload the data using the bulk upload process.  A few questions:
1)  Is it possible to calculate a time-stamp as a covariate for each plot-level data?  Is calculating the median time for each plot something that Matt could easily do? 
2) I don't see a citation for Matt Maimaitiyiming in BETYdb.  Should I create one for Matt 2016, or use the existing citation for Pedro 2016? 
3) There's an existing Method called 'Maricopa Phenotractor'.  Has the bulk upload process been revised so that it's now possible to link a Method with a data-set during upload?  

I made a new folder 'tractor data 2016 for upload' that has the formatted data, 4 separate spreadsheets to make it easy to use the bulk upload process:  https://drive.google.com/drive/folders/0B1OR5-zeXagnejZqN1ZVV05zY2M

dlebauer, 2017-09-23:
> Is it possible to calculate a time-stamp as a covariate for each plot-level data? Is calculating the median time for each plot something that Matt could easily do?

The only files I have are from Sept 1: sites/ua-mac/raw_data/phenotractor/2016-09-01 which doesn't have a time field, and these https://drive.google.com/drive/u/0/folders/0By_PDCcY5g2JUlpJSEcwbFQxRk0

@Mamatemenrs do you still have the original files? Can you please share them?

> I don't see a citation for Matt Maimaitiyiming in BETYdb. Should I create one for Matt 2016, or use the existing citation for Pedro 2016?

We should use Pedro ... or a new citation with both Wasit and Pedro. The key is to make sure Matt has a chance to participate in any publications. At that point we can update the citation that will more accurately reflect contributors than the 'unpublished data' citations we use in the meantime.

> There's an existing Method called 'Maricopa Phenotractor'. Has the bulk upload process been revised so that it's now possible to link a Method with a data-set during upload?

Sorry, this feature is still in queue (because it is straigtforward for me to do it manually) ... the feature is https://github.com/PecanProject/bety/issues/470. You can keep track of these in this (or other relevant open) issue, in new issues, or this big overarching issue: https://github.com/terraref/reference-data/issues/189

dlebauer, 2017-09-23:
Actually, your original issue here: https://github.com/terraref/reference-data/issues/159 is probably the best place

NewcombMaria, 2017-09-23:
If it turns out we need to go back to the original files to get the time data, Pedro and John Heun have all the un-processed data.  Matt must have set up a nice pipeline to get to geo-referenced plot-level statistics.  @Mamatemenrs nice work!  If you can add the GPS time-stamp data to your pipeline that would be an important covariate to relate NDVI and surface/canopy temp to environmental data like PAR and air temp.

NewcombMaria, 2017-09-28:
By email from Pedro:
All,
I asked John H. to compile the data files we spoke yesterday (i.e. 2016 tractor runs, and 2017 fixed-point canopy temperature). This link contains both sets:
 
https://arizona.box.com/s/m3h7g93q9wri7odx57cwz12ixlice1jc
 
Couple of comments: there were actually five tractor runs in 2016, Matt processed only four, I think we only had four to share with him at that time. For the canopy temperature folder, there are four files (one for each location) Maria knows the field locations of these sensors. The headers are self-explanatory, there is date and time of the day columns, then canopy temp is in column E (Avg target temp(C)). Let me know if you have questions, 
Pedro

NewcombMaria, 2017-09-29:
Updates by email from Pedro and Matt: 

_Maria brought to our attention that she logged a tractor run in the 2017 season. Indeed there was a tractor run on May 19 just before we moved this platform to California. John uploaded the file to the folder in Box indicated below. This is an excel file with the offsets in a separate tab. Let me know if you have questions,
Pedro_

_I am attaching the plot level tractor data updated with timestamps.  Please let me know if I can be of further assistance. 
Best, 
Matt_ 

I will give a first-run to upload to betydb the data Matt sent with date-time included in the next few days.

NewcombMaria, 2017-10-02:
Matt sent a revised spreadsheet that includes time data, along with a description of methods that includes a photo of the tractor set-up (both added to the directory https://drive.google.com/drive/folders/0B1OR5-zeXagnejZqN1ZVV05zY2M ).  I've added time data to the formatted files in the same directory (4 files each for a single date).  As a first-level quality check I sorted the data by time and also by 'pass' and the sorting arrangement doesn't completely make sense for a N-S or S-N run by the tractor.  The data will need some review before uploading.

NewcombMaria, 2017-10-03:
Update from Matt by email: 
_The plot level timestamp for some plots is the average of one tractor path (only to north or south), while for some other plots it is an average of two tractor paths (north and south or vice versa). This is due to the three different sensors installed on the tractor. Please also see the attached figure. 
Thanks. 
Matt_ 

[image]

NewcombMaria, 2017-10-03:
The example of '11-7' shown in Matt's image raises a question about calculating plot-level statistics when multiple runs at different time-points are required to cover the plot.  With only 7 or 8 minutes difference it may not be that important.  The most accurate measurement instead of avg canopy temp at the avg time might be the single-pass temp differential (difference between air temp and leaf temp) averaged over the plot area.

NewcombMaria, 2017-10-04:
@dlebauer it looks like you or Scott will need to upload the revised and reformatted tractor data-set.  I just tried to upload under the citation Andrade Pedro 2016 and run into 2 problems:
1)  I get the error report: Data Value Errors
* Site is inconsistent with citation in these rows: 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11 ... 
2) The column for local_datetime is ignored, as seen earlier - closed issue #162 

The data are ready to go, see 4 separate csv files (Sorghum_season2_tractordata ending in _a, _b, _c, _d) at: https://drive.google.com/drive/folders/0B1OR5-zeXagnejZqN1ZVV05zY2M 
I've modified the Method called 'Maricopa Phenotractor' which should be linked with the variables NDVI, canopy_height, surface_temperature (all defined already), and with local_datetime.  

Let me know if there are questions or if other steps are needed on my end.  Thanks.

NewcombMaria, 2017-10-26:
I changed the name of this issue to include the single date from 2017 season4 with tractor-platform data.  Jeff White agreed recently to summarize the data from 2017-05-19 for plot-level statistics that can be uploaded to betydb before the Nov 15 data release.

gsrohde, 2017-11-10:
@NewcombMaria, @dlebauer Did one of you upload this data, or am I to do it?

NewcombMaria, 2017-11-10:
@gsrohde I did not upload because I ran 2 problems:  1) time data, and 2) when I used Pedro's citation (Andrade Pedro 2016) I ran into the problem with and error message: Site is inconsistent with citation in these rows: 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11 ... 

Let me know if you need any additional info from me.

dlebauer, 2017-11-10:
@gsrohde although it would be better if citations_sites records were automagically created in this case, I am also wondering if you could show me how to combine the following queries into insert statements

```sql
/* query 1 */
select id from citations where author = 'Andrade, Pedro';
/* query 2 */
select id from sites where sitename like 'MAC Field Scanner%' and id not in (select distinct site_id from citations_sites);
```

Into something like 

```sql
insert into citations_sites (citation_id, site_id) values (query 1, query 2);
```

gsrohde, 2017-11-11:
@dlebauer Do you want to insert all combinations of citation_id and site_id where the citation_id is in the set of values returned by query 1 and the site_id is in the set of values returned by query 2?  If so, you can do this:
```
INSERT INTO citations_sites (citation_id, site_id) SELECT c.id, s.id FROM (query 1) AS c CROSS JOIN (query 2) AS s;
```

dlebauer, 2017-11-11:
Yep I was looking for cross join but didn't know the name. Thanks!

dlebauer, 2017-11-11:
done

```sql
INSERT INTO citations_sites (citation_id, site_id) SELECT c.id, s.id FROM (select id from citations where author = 'Andrade, Pedro') AS c CROSS JOIN (select id from sites where 
    sitename like 'MAC Field Scanner%'
    and id not in (select distinct site_id from citations_sites where 
          citation_id not in (select id from citations where author = 'Andrade, Pedro'))
) AS s;
```

gsrohde, 2017-11-13:
@NewcombMaria The two problems you cited should no longer stand in the way of your uploading these: The API handles time data, and David added the needed citation-site associations.

Of course, as you probably realize, the CSV files need a lot of fixing up before they are ready to be uploaded.  If I were to upload these, I would have to ask you many questions.