terraref/reference-data #193: Renumber rows with id 6000000000.

gsrohde, 2017-09-21:
Because the migrations for some of the new tables set the initial id values for some sequences to be `hostid x 10e9` instead of `hostid x 10e9 + 1`, there are rows in the `experiments`, `experiments_sites`, and `experiments_treatments` tables on bety6 having id number 6000000000.  But these rows don't propagate to other machines when a sync is done.  This results in some foreign-key violations on the importing machines.  This is because the sync script turns off constraint checking during the synchronization process.

These rows should be fixed by running the following statements on the bety6 bety database:

```
UPDATE experiments SET id = nextval('experiments_id_seq') WHERE id = 6000000000;
UPDATE experiments_sites SET id = nextval('experiments_sites_id_seq') WHERE id = 6000000000;
UPDATE experiments_treatments SET id = nextval('experiments_treatments_id_seq') WHERE id = 6000000000;
```

In the case of the experiments table update, the `ON UPDATE CASCADE` clauses in the foreign-key constraints of the referring tables ensure that foreign keys referring to the experiments table are correctly updated.

I haven't found any other tables on bety6 having rows with id 6000000000.  Other machines should be checked as well.  If any table has a row with an id that is an integral multiple of 1000000000, it should be updated.  For empty tables, the value of the sequence objects for those tables should be checked to ensure that numbering will start with the correct value.

* [ ] Fix `experiments` and related tables on bety6.
* [ ] Check for potential problems in other tables and on other machines.

dlebauer, 2017-10-05:
Is this waiting on anything or can you go ahead and run the corrections?

Is there an issue to correct the indices that started at 1?

gsrohde, 2017-10-05:
Closing.  Duplicated by #195 (which also deals with the indices starting at 1).

closed 2017-10-05 by gsrohde