terraref/reference-data #194: Check plot boundaries from all seasons and alignment

dlebauer, 2017-09-28:
Following https://github.com/terraref/computing-pipeline/issues/361, I suspect that we need to replace all of the plot definitions with the ones that Rick is using with his shape files (these capture the center two rows of four-row plots and represent where the hand measurements were taken). But before doing this, need to confirm that the gantry and UAV have consistent geospatial coordinates by looking at those plots overlaid on both UAV and gantry stitched images.

### Completion Criteria 

Need to confirm that we have the correct and consistent plot definitions

1. assemble UAV and full field RGB mosaic thumbnails from early in each of season (between germination and canopy closure) for seasons 1, 2, and 4, approximately within 2 weeks of 
   * season 1 2016-06-01
   * season 2 2016-08-31
   * season 4 2017-06-01
2. overlay plot definitions currently in BETYdb
   * make these maps available (e.g. publish in arcgis online)
3. revise plot definitions in BETYdb if necessary.

max-zilla, 2017-10-02:
in /sites/ua-mac/Level_1/fullfield...

- 2017-06-01 completed on Friday.
- 2016-06-12 completed on Friday, but it turns out it was not a complete scan.
- 2016-06-13 processed over the weekend and I queued the full field yesterday, it should finish sometime this morning.
- 2016-08-23 same, currently processing on 2nd extractor. Should finish sometime this morning.

In these directories, look at "...rgb_thumb.tif" file.

-Max

dlebauer, 2017-10-02:
@max-zilla this is the north end of the field for 2016-08-23 (the 2016-06-12 looks similar). There seem to be a lot of missing elements. Is this to be expected?

[image]

dlebauer, 2017-10-02:
As described in https://github.com/terraref/computing-pipeline/issues/361#issuecomment-333680688, Season 2 is OK (consistent w/ plots from @rickw-ward and @anfrench): 

[image]

dlebauer, 2017-10-02:
This is what the season 2 plots look like when overlain on plot boundaries: 

[image]

dlebauer, 2017-10-03:
## Season 1

Appears that there may be a problem: the plot boundaries appear to go down the center of the rows (e.g. along the line where seeds were planted)

[image]

Note that for each plot (in red) there are E and W subplots (orange)

An easy fix would be to shift the plots east or west ... but which way, and how far?

rickw-ward, 2017-10-03:
need to see edge of field (east or west) to know direction/distance of shift that is needed.

NewcombMaria, 2017-10-03:
Regarding how far, row width is 76.2 cm, so shifting half-row or 38.1 cm to the left looks like it would capture a single genotype/cultivar (left 2 rows look like the same genotype).

dlebauer, 2017-10-03:
@rickw-ward I don't have any images that extend to the edge of the field for Season 1, but this is what the northern end of the field looks like

[image]

rickw-ward, 2017-10-03:
Below is screen capture of NDVI ortho from june 5 2016, north end of gantry.  @NewcombMaria 's suggestion to shift left 38.1 cm looks right.

[image]

dlebauer, 2017-10-04:
## Season 4

Looks good. 

subplots E (red) and W (orange) shown 

[image]

difficult to see but subplots from Rick and Andy's shapefile are in grey

[image]

Database also has full plots (E+W combined, here with Rick & Andy's subplots still in grey) thus: 
[image]

dlebauer, 2017-10-04:
This should fix season 1 ... will check tomorrow

```sql
update sites set geometry = st_translate(geometry, -0.00000818099, 0) where sitename like 'MAC Field Scanner Season 1%'
```

dlebauer, 2017-10-05:
## Season 1 appears fixed

Before: plot borders down the row:

[image]

After:

[image]

closed 2017-10-05 by dlebauer

dlebauer, 2017-10-09:
Here are the season 1 plots overlain on the 2016-06-05 NDVI map 

[image]

rickw-ward, 2017-10-09:
@dlebauer and @NewcombMaria good to see.   Can you share json or SHP file(s) for both seasons 1 and 2, now that they align for both of us? Sorry if this is redundant and they are already in my hands.

dlebauer, 2017-10-09:
I  uploaded the shapefiles  to https://github.com/terraref/computing-pipeline/issues/361#issuecomment-335210 files

And here is ththe direct link: https://github.com/terraref/computing-pipeline/files/1369204/season124shpfiles.zip

dlebauer, 2018-02-02:
Update: Season 4 had reversed E and W and was fixed here: https://github.com/terraref/computing-pipeline/issues/408#issuecomment-362716188