terraref/reference-data #195: id number updates needed due to mis-set id sequence values

gsrohde, 2017-10-05:
Completion checklist
* [ ] Correct experiments and related tables on bety6
* [ ] Correct sites_cultivars on bety6
* [ ] Figure out how to set sequence objects and make needed corrections on other source machines and client machines
---

The initial value for the sequence objects for some of the tables in the terra-ref database was not set correctly.  In general, the initial value returned by the object should be 6000000001.

## Cases where the initial value was set to 6000000000.

The sequence objects for the tables `experiments`, `experiments_sites`, and `experiments_treatments` initially returned a value of 6000000000.  Therefore each of these tables has a row with id 6000000000.  But the database dump script used to copy data from machine 6 (the main terra-ref database machine) to user copies dumps only rows having ids in the range 6000000001–7000000000.  Thus they won't get the first-inserted row of these tables.

### Fix

These commands should fix the tables:
```
UPDATE experiments SET id = (SELECT nextval('experiments_id_seq')) WHERE id = 6000000000;
UPDATE experiments_sites SET id = (SELECT nextval('experiments_id_seq')) WHERE id = 6000000000;
UPDATE experiments_treatments SET id = (SELECT nextval('experiments_id_seq')) WHERE id = 6000000000;
```
 Note: The fact that the foreign-key constraints "fk_experiments_sites_experiments" and "fk_experiments_treatments_experiments" both cascade on update means that the reference `experiment_id` in these tables will be updated appropriately when the experiments table is updated.

## Cases where the initial value was set to 1.

The sequence object for the table `sites_cultivars` initially returned a value of 1.  This means all rows in the table have id numbers in the range 1–1000000000 instead of 6000000001–7000000000.  

### Fix

Since `sites_cultivars` is empty in the `ebi_production` database, this means all of these rows originated on bety6.  Therefore we can simply update these rows to move the id numbers into the bety6 range.  These commands should work:

```
SELECT setval('sites_cultivars_id_seq', 6000000001, 'false');
UPDATE sites_cultivars SET id = nextval('sites_cultivars_id_seq');
```

gsrohde, 2017-10-05:
@dlebauer Does your assigning yourself this mean you'll run the updates?

dlebauer, 2017-10-05:
I assigned it to myself to take a look - But I can go ahead and update

for the first set,  this line gave an error:

```sql
UPDATE experiments_sites SET id = nextval('experiments_id_seq') WHERE id = 6000000000;
```
> [Err] ERROR:  duplicate key value violates unique constraint "experiments_sites_pkey"
DETAIL:  Key (id)=(6000000023) already exists.

The sites_cultivar update worked.

dlebauer, 2017-10-05:
Other than terra-mepp, I don't think that the PEcAn community is using the sites_cultivars or experiments tables

gsrohde, 2017-10-05:
OK, I updated `experiments_sites` (you shouldn't have checked off item 1 before I did!).  I had referenced the wrong sequence objects in the update statements, which I've since corrected.  I'll check terra-mepp for needed updates.

closed 2020-02-13 by dlebauer