terraref/reference-data #196: How to encode MAC Season 4 drought experiments

dlebauer, 2017-10-18:
Breaking out this issue from https://github.com/terraref/reference-data/issues/174#issuecomment-334807104 

from @NewcombMaria 
... the Managements I entered for 2017-08-01 and 2017-08-15 describe the establishment and reversal of 'water-deficit' and 'well-watered' blocks as part of the Field Campaign Stress Effect on Photosynthesis. I entered them under the Treatment 'Control BAP 2017'. Is this correct? Or should the month of August be split into 2 separate treatments for the time intervals August 1-14 and August 15-30?

dlebauer, 2017-10-18:
@NewcombMaria regarding the irrigation / water deficit treatment. It looks like there were two treatments, one that was put under water stress at Day 103 (Aug 1) and the other at day 116 (Aug 15).

In this case, I would create two treatments, for example _BAP 2017 drought 103 days after planting" and "BAP 2017 with drought at 117 days after planting" 

Then we can link all of the existing managments (planting, tillage, fertilization, etc) to both of these and then add managements called 'irrigation' with level = 0 and date Aug 01 and link it just to the 103 dap drought treatment. Another management with irrigation level = 0 and date Aug 15 and link it to the 116 dap drought treatment. In addition, we can add managements with the same initial irrigation rates on these dates and attach them to the opposite treatment. 

So the irrigation timeline for 103 dap would look like 

| Date       | Management   | Level | Units | Description |
|------------|--------------|-------|-------|-------------|
| ... |    |       |       |             |
| 2017-04-20 |planting   |       |       |             |
| ... |    |       |       |             |
| 2017-07-30 | irrigation   |  20    | mm/d      |             |
| 2017-08-01 | irrigation   |   0    | mm / d       |             |
| 2017-08-15 | irrigation   | 20      | mm /d       |             |
| 2017-08-30 | irrigation   | 20      | mm / d      |             |
| 2017-09-11 | harvest      |       |       |             |
|...  | |       |       |             |

So the irrigation timeline for 116 dap would have all of the same managements except the 8-01 and 08-15 dates:

| Date       | Management   | Level | Units | Description |
|------------|--------------|-------|-------|-------------|
|...  | |       |       |             |
| **2017-08-01** | irrigation   |   20    | mm / d       |             |
| **2017-08-15** | irrigation   | 0      | mm /d      |             |
| 2017-08-30 | irrigation   | 20      | mm / d      |             |
|...  | |       |       |             |

The description / treatments can point to the raw irrigation data here: https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?stream_id=3212

Which look like this:

[image]

(code here: https://gist.github.com/dlebauer/ff5826d9528b720bbd7a5129c25ad319)

---

However, this has uncovered an invalid assumption in our irrigation data stream: we have been assuming that the area irrigated is constant, at 4000 m2 in order to convert from galons / d to CF standard units of kg / m2 / s (= mm / s). 

The code is here: https://github.com/terraref/extractors-environmental/blob/master/irrigation_datparser/parser.py#L10

So, we need to revisit this calculation. One way would be to create a new treatment named 'irrigated_area', and the value would be valid from the date of the treatment until a new value occurs.

@NewcombMaria could you or @Andrade-Pedro provide the values for irrigated area: is 4000 m2 a sensible default? Then, do you know what fraction of the field was in the drought treatments?

_NB: we won't have time to fix this before Nov, but can make it high priority shortly thereafter_

NewcombMaria, 2017-10-19:
@dlebauer the irrigation timeline is correct (one part of the field at level 0 from 103 to 116 DAP and returning to level 20 after 116 DAP, while the opposite section of the field remains at level 20 until 116 DAP then drops to 0 .  Thanks!  

- [ ] I'll talk with Pedro and we'll figure out the proportion of the field (m2) in the water-restricted area for each of the time periods.  Knowing the area (proportion of the field) will allow for correction of the irrigation data stream.  

- [ ] Do we still need to link specific plots (sites) to the water-deficit time periods? If it's helpful I can share a spreadsheet with each site (plot) described by water-deficit or well-watered for each time period (103 - 116 DAP and 116 - 131 DAP).  Let me know how to geo-reference the water-deficit areas.

NewcombMaria, 2017-10-19:
@dlebauer here are the values for area irrigated by time that @Andrade-Pedro and I worked out.  Sorghum season4 irrigated area was slightly larger compared to previous seasons because we added an additional buffer row in the narrow spaces on the West and East edges (outside of the scan-able area but inside the rails, and at narrower row spacing).  I don't know the history of when/how the value 4000 m2 irrigated area originated.  In previous seasons I think the irrigated area would have been 5267 m2, (216 m x 24.384 m)but perhaps there's an explanation for 4000 m2 that I don't know about. 

Full-field area (including buffer rows) that was irrigated from the start of the season to August 1 (103 dap):  5486.4 m2 (=216 m x 25.4 m)

Well-watered treatment zones from August 1 to 15 (103 to 116 dap):  2522.88 m2 (=216 m x 11.68 m) in total, combined areas of non-contiguous blocks 

Well-watered treatment zones from August 15 - 30 (116 to 131 dap):  3181.68 m2 (=216 m x 14.73 m), again in total as the combined areas of non-contiguous blocks

dlebauer, 2017-10-19:
> but perhaps there's an explanation for 4000 m2 that I don't know about.

I've always thought the field dimensions were 200 x 20m but I can see that this was just an approximation. 

* [ ] If you provide the list of plot names in each treatment, I can match the plots with specific treatments in the database. Then in future uploads it will be easiest if each file contains both a site and treatment column.

Do you or Pedro have polygons representing the irrigation zones? I am wondering if 1) computing the area is gis operation that could be automated and 2) if we can record irrigation zones as 'sites' so that we can easily query all plots within an irrigation zone and compute areas irrigated.

NewcombMaria, 2017-10-19:
@dlebauer I added 2 spreadsheets to the directory https://drive.google.com/drive/folders/0B1OR5-zeXagnUGdLb0dVUFRYc1k  
with the site names at the plot-level matched to WW (well-watered or irrigated) or D (dry or not irrigated).  These irrigation treatments only are associated with data collected in August, not for any data collected from April 20 to July 31.  One spreadsheet is for the irrigation blocks from 2017-08-01 to 2017-08-15; the other spreadsheet has the irrigation blocks from 2017-08-16 to 2017-08-30.  (Could have merged them into one spreadsheet with different columns but this is how I have the treatments for now).

NewcombMaria, 2017-10-19:
We don't have GIS polygons for the irrigation blocks in August.  The areas in previous comments are accurate, based on row widths and range lengths.  We could create polygons for the experimental and border plots based on the descriptions in the Management list:
https://terraref.ncsa.illinois.edu/bety/managements/6000000079  
https://terraref.ncsa.illinois.edu/bety/managements/6000000080

dlebauer, 2017-11-09:
@NewcombMaria indeed we should be able to create polygons for a given management thus:

### water-deficit blocks with restricted irrigation starting Aug 01
```sql
select st_astext(st_union(geometry)) as geometry 
     from sites 
     where sitename ~ 'Column (5|6|7|12|13|14|15)$' 
              and sitename like 'MAC Field Scanner Season 4%'
```

```
 MULTIPOLYGON Z (((-111.974875650957 33.0745634337264 0,-111.97485928187 33.0745634390379 0,-111.974842912784 33.0745634443472 0,-111.974826543697 33.0745634496543 0,-111.97481017461 33.0745634549593 0,-111.974810160268 33.0745318872505 0,-111.974826529349 33.0745318819456 0,-111.97484289843 33.0745318766385 0,-111.974859267404 33.0745318713302 0,-111.974875636484 33.0745318660188 0,-111.974875650957 33.0745634337264 0)),((-111.974990234564 33.0745633965761 0,-111.974973865478 33.0745634019027 0,-111.97495749639 33.0745634071368 0,-111.974941127304 33.074563412459 0,-111.974941112808 33.0745318447519 0,-111.974957481888 33.0745318394298 0,-111.974973850969 33.0745318341055 0,-111.974990220049 33.0745318287791 0,-111.974990234564 33.0745633965761 0)),((-111.974990251052 33.0745993386599 0,-111.974973881959 33.0745993439865 0,-111.974957512865 33.074599349311 0,-111.974941143772 33.0745993546333 0,-111.974941129275 33.0745677868362 0,-111.974957498363 33.0745677816041 0,-111.97497386745 33.0745677762798 0,-111.974990236538 33.0745677709532 0,-111.974990251052 33.0745993386599 0)),((-111.974875667398 33.074599375811 0,-111.974859298305 33.0745993811226 0,-111.974842929211 33.074599386432 0,-111.974826560118 33.0745993917393 0,-111.974810191024 33.0745993970445 0,-111.974810176682 33.0745678293358 0,-111.974826545769 33.0745678240308 0,-111.97484291475 33.0745678187244 0,-111.974859283838 33.0745678134151 0,-111.974875652925 33.0745678081036 0,-111.974875667398 33.074599375811 0)),((-111.974990283921 33.0746712228275 0,-111.974973914922 33.0746712281536 0,-111.974957545815 33.0746712334783 0,-111.974941176709 33.0746712388009 0,-111.974941162212 33.0746396710943 0,-111.974957531312 33.0746396657718 0,-111.974973900413 33.0746396604472 0,-111.974990269513 33.0746396551204 0,-111.974990283921 33.0746712228275 0)),((-111.974875700282 33.0746712600698 0,-111.974859331176 33.0746712653816 0,-111.974842962069 33.0746712706913 0,-111.974826592961 33.0746712759086 0,-111.974810223854 33.074671281214 0,-111.974810209404 33.0746397135066 0,-111.974826578505 33.0746397082013 0,-111.974842947606 33.0746397028938 0,-111.974859316707 33.0746396975843 0,-111.974875685809 33.0746396923627 0,-111.974875700282 33.0746712600698 0)),((-111.97499026754 33.0746352807434 0,-111.97497389844 33.0746352860701 0,-111.97495752934 33.0746352913947 0,-111.97494116024 33.0746352967172 0,-111.974941145744 33.0746037290104 0,-111.974957514838 33.0746037236881 0,-111.974973883931 33.0746037183636 0,-111.974990253025 33.0746037130369 0,-111.97499026754 33.0746352807434 0)),((-111.97487568384 33.0746353178954 0,-111.97485931474 33.0746353232071 0,-111.974842945639 33.0746353285167 0,-111.974826576539 33.0746353338241 0,-111.974810207439 33.0746353391294 0,-111.974810193096 33.0746037714209 0,-111.974826562083 33.0746037661165 0,-111.974842931178 33.0746037608092 0,-111.974859300272 33.0746037554998 0,-111.974875669366 33.0746037501882 0,-111.97487568384 33.0746353178954 0)),((-111.9749903169 33.0747431070837 0,-111.97497394778 33.0747431124109 0,-111.974957578661 33.0747431177358 0,-111.97494120954 33.0747431229685 0,-111.974941195149 33.0747115552614 0,-111.974957564263 33.0747115499387 0,-111.974973933377 33.0747115446138 0,-111.974990302492 33.074711539377 0,-111.9749903169 33.0747431070837 0)),((-111.974875733167 33.0747431442375 0,-111.974859364047 33.0747431495496 0,-111.974842994927 33.0747431548596 0,-111.974826625807 33.0747431601674 0,-111.974810256687 33.074743165473 0,-111.974810242236 33.0747115977659 0,-111.97482661135 33.0747115924603 0,-111.974842980464 33.0747115871527 0,-111.974859349579 33.0747115818428 0,-111.974875718693 33.0747115765309 0,-111.974875733167 33.0747431442375 0)),((-111.97499034988 33.0748149912489 0,-111.974973980747 33.0748149965763 0,-111.974957611614 33.0748150019015 0,-111.97494124248 33.0748150072246 0,-111.974941228089 33.0747834395179 0,-111.974957597216 33.0747834341949 0,-111.974973966343 33.0747834288698 0,-111.974990335471 33.0747834235425 0,-111.97499034988 33.0748149912489 0)),((-111.974875766055 33.0748150284947 0,-111.97485939692 33.0748150337168 0,-111.974843027787 33.074815039027 0,-111.974826658653 33.0748150443351 0,-111.97481028952 33.074815049641 0,-111.974810275068 33.0747834819342 0,-111.974826644196 33.0747834766284 0,-111.974843013323 33.0747834713205 0,-111.974859382451 33.0747834660104 0,-111.974875751579 33.0747834606982 0,-111.974875766055 33.0748150284947 0)),((-111.97487574961 33.0747790863211 0,-111.974859380484 33.0747790916333 0,-111.974843011357 33.0747790969434 0,-111.97482664223 33.0747791022513 0,-111.974810273103 33.0747791075571 0,-111.974810258652 33.0747475398502 0,-111.974826627773 33.0747475345445 0,-111.974842996894 33.0747475292367 0,-111.974859366015 33.0747475239267 0,-111.974875735135 33.0747475186146 0,-111.97487574961 33.0747790863211 0)),((-111.97499033339 33.0747790491664 0,-111.974973964263 33.0747790544937 0,-111.974957595137 33.0747790598188 0,-111.97494122601 33.0747790651418 0,-111.974941211619 33.0747474974349 0,-111.97495758074 33.074747492112 0,-111.97497394986 33.074747486787 0,-111.974990318 [... cut ...] 57816651 0,-111.974892597712 33.0758257763471 0,-111.974908967033 33.0758257710269 0,-111.974925336353 33.0758257657046 0,-111.974925350749 33.0758573334062 0)),((-111.975039936032 33.0758572961792 0,-111.975023566705 33.0758573014265 0,-111.975007197379 33.0758573067618 0,-111.974990828053 33.075857312095 0,-111.974990813527 33.0758257443947 0,-111.975007182847 33.0758257390616 0,-111.975023552167 33.0758257337264 0,-111.975039921487 33.075825728389 0,-111.975039936032 33.0758572961792 0)))
```

NewcombMaria, 2017-11-12:
@dlebauer if I'm understanding correctly, I should create 4 new treatments?  So far there is one treatment for Season 4 named as 'Control BAP 2017' which is irrigated for optimal growth and was implemented from 2017-04-20 to 2017-07-31.  Should I create 4 new treatments?:

Irrigation=0 BAP 2017-08-01 to 2017-08-15 
Irrigation=1 BAP 2017-08-01 to 2017-08-15
Irrigation=0 BAP 2017-08-16 to 2017-08-30 
Irrigation=1 BAP 2017-08-16 to 2017-08-30 

See for example the treatment column (L) in the spreadsheet with August data from the MultispeQ device at the Google directory 
https://drive.google.com/drive/folders/1MVkQpJ8f_gCHZP17j63tXaAGCT1R5tAO 
Will this work for defining the August irrigation treatments?

dlebauer, 2017-11-12:
I think there should be two treatments that cover two cases: one begins
drought Aug 1 and the other begins drought Aug 15,

All plots fall into one of these two, correct?

NewcombMaria, 2017-11-12:
With two new treatments [1) drought Aug 1-15 and 2) drought Aug 16-30], then do the non-drought plots that were irrigated during those times fall under 'Control BAP 2017' treatment?

dlebauer, 2017-11-12:
Yes.

dlebauer, 2017-11-12:
Were there plots that were never droughted?

NewcombMaria, 2017-11-12:
All plots were drought-stressed either during Aug 1 to 15 or Aug 16 to 30.  Plots that were drought-stressed in early August and then returned to normal irrigation latter August were considered to be in 'recovery' from water-deficit stress.  This Recovery status could be considered a 'Stress Recovery' treatment different from continuous well-watered status.  The goal was to compare accessions for abilities to recover after drought.

dlebauer, 2017-11-12:
Then just two treatments. They apply for the whole season. You can rename
the control.

NewcombMaria, 2017-11-12:
Two treatments total, for the BAP 2017 (season 4):
'Control BAP 2107' = well-watered status irrigated for optimal growing conditions
'Irrigation = 0 BAP 2017' = zones with no irrigation during times in August to impose water-deficit stress 

Something like that?

dlebauer, 2017-11-13:
Sorry for the confusion -  all of the treatments will apply for the entire
season.

if as stated above

> All plots were drought-stressed either during Aug 1 to 15 or Aug 16 to 30.

there would be one treatment for plots that were drought stressed Aug 1-15
and another for plots that were drought stressed Aug 16-30.

These can be named 'BAP 2017 with drought Aug 1-15' and 'BAP 2017 with
drought Aug 16-30'

The timing and amounts of irrigation will be encoded as managements; I can
add these relationships if you rename the existing 'Control BAP 2017' and
create a new treatment to use for uploading.

NewcombMaria, 2017-11-13:
@dlebauer thanks.  I think I get it now.  The 2 treatments will be defined as 2 areas of the field from planting to harvest, one area with no irrigation the first half August and the second area with no irrigation the second half August.  All of the data collected in May, June, July already uploaded under one treatment (Control BAP 2017) will need to be revised to distinguish the sites (plots) by their August irrigation zones?  Could we talk briefly on Monday to make sure I'm understanding correctly?

NewcombMaria, 2017-11-14:
@dlebauer I renamed and revised the treatment 'Control BAP 2017', now named 'BAP 2017, water-deficit stress Aug 1-14'.  See https://terraref.ncsa.illinois.edu/bety/treatments/6000000024 

I also created a new treatment 'BAP 2017, water-deficit stress Aug 15-30'.  See https://terraref.ncsa.illinois.edu/bety/treatments/6000000026 

Right now there are no managements associated with the newly created treatment.  Let me know if I need to revise or add anything else.  Thanks.

dlebauer, 2018-01-25:
@NewcombMaria or @Andrade-Pedro can one of you provide a map with an outline of the irrigation areas?

NewcombMaria, 2018-01-26:
@dlebauer these are images showing the water-deficit blocks during the August drought treatments in August (not to scale).  The areas outlined in red were water-limited (no irrigation) during the dates shown in the maps.  Areas not outlined in red (all other areas of the field) were well-watered during the dates shown in the maps.  The entire field area (216 m x 25.4 m) was well-watered from planting to August 1.  Let me know if you have questions. 

[image]

[image]

dlebauer, 2018-01-26:
@NewcombMaria thanks. For the puposes of figuring out how to handle these:

* to confirm - were the border rows were always irrigated?
* how many zones are there? It looks like there are four zones plus the E/W border rows. Is this correct?
* are these zones somehow fixed?
    * will future irrigation treatments use these same zones?
* what names (if any) do you have for referring to the different zones? 
   * otherwise I'll make something up like Irrigation Zone 1, Irrigation Zone 2, ...

dlebauer, 2018-01-26:
On closer look, it appears that the edges aren't actually border rows ... do you know how wide they are?

dlebauer, 2018-01-26:
In https://github.com/terraref/reference-data/issues/196#issuecomment-338015666 you say that the total area of the irrigated area is 216 m x 25.4 m = 5486m2, but when I combine all of our plots I get 214.7 x 24.4 = 5249m2

This is a view of the the SW and NW corners of the field with the 214.7 x 24.4 rectangle that includes all of the plots in orange. 

[image]

If I add 0.5, to the E and W and 0.65 m to the E and W I would get the correct irrigated area ... does that seem reasonable?

NewcombMaria, 2018-01-27:
@dlebauer he full-field area (including buffer rows) that was irrigated from the start of the season to August 1 (103 dap) is 5486.4 m2 (=216 m x 25.4 m).  This is correct.  There are 54 planted/irrigated ranges and each ranges is exactly 4.0 m in length in the N-S axis.  54 x 4 m = 216 m.  The planted/irrigated width is 25.4 m (32 rows at 0.762 m spacing and 2 additional outer border rows on E and W edges planted at different spacing by a different planter at 0.508 m spacing = 25.4 m total).  These values are accurate for the exact area of the field that is irrigated.  

Well-watered treatment zones from August 1 to 15 (103 to 116 dap): 2522.88 m2 (=216 m x 11.68 m) in total, combined areas of non-contiguous blocks

Well-watered treatment zones from August 15 - 30 (116 to 131 dap): 3181.68 m2 (=216 m x 14.73 m), again in total as the combined areas of non-contiguous blocks 

Does that help?  Let me know if my explanation doesn't make sense, or send additional questions.

NewcombMaria, 2018-01-27:
@dlebauer  these irrigation zones are not fixed.  They were planned mid-season for the purpose of the field campaign related to water-deficit stress effects on photosynthesis in Aug 2017.  They were blocks implemented after mid-season discussions with those who helped plan the August field campaign.

NewcombMaria, 2018-01-27:
@dlebauer I'm rethinking how you could have come up with 214.7 m in the N-S axis.  The plot on the northern end (range 54) include an alley, length 0.5 m.  Since all other plots are associated with an alley, we've been treating plots on the north end of the field the same.  That changes the field length to **215.2 m**, which is in fact the most accurate field length.  We learned after planting that even though the precision planter was set for plot length 4.0 m, it turned out that we did measure a systematic slight difference in each plot length after planting due to a planter off-set rounding error.  Each plot was actually 3.985 m instead of 4 m, and this difference adds up to 0.8 m across the 54 ranges.  

This means the most accurate irrigated field area and treatment 1 and 2 irrigated areas are as follows: 

Full field irrigated area from the start of the season to August 1 (103 dap) is **5466.1 m2** (=215.2 m x 25.4 m). 

Well-watered treatment zones from August 1 to 15 (103 to 116 dap): **2513.5 m2** (=215.2 m x 11.68 m) in total, combined areas of non-contiguous blocks

Well-watered treatment zones from August 15 - 30 (116 to 131 dap): **3169.9 m2** (=215.2 m x 14.73 m), again in total as the combined areas of non-contiguous blocks

In reality, the drip lines extend beyond the planted area on both the south and north ends of the field to connect to the lateral submain and to the termination point at the end of the drip line.  The quantified areas are the planted areas that are irrigated.  I hope this is helpful and makes sense.

dlebauer, 2020-03-07:
Set treatments thus: 

```
begin;
update traits set treatment_id = 6000000024 where site_id in (
select id from sites where sitename ~ 'Column (5|6|7|12|13|14|15|16)$' 
              and sitename like 'MAC Field Scanner Season 4%') and treatment_id is NULL;
commit;
begin;
update traits set treatment_id = 6000000026 where site_id in (
select id from sites where sitename ~ 'Column (1|2|3|4|9|10|11)$' 
              and sitename like 'MAC Field Scanner Season 4%') and treatment_id is NULL;
commit;
```

closed 2020-03-07 by dlebauer