terraref/reference-data #198: How to connect new trait variable names to required covariates

NewcombMaria, 2017-10-26:
@dlebauer or @gsrohde I'm trying to upload to betydb a small data-set of stalk diameter field measurements.  The data set has 2 newly defined trait variable names that both need to connect to the required covariate 'leaf_position'.  When I try to upload the data file I get the error message: 
**Field List Errors
* The following heading names correspond to covariates but the heading contains no corresponding trait variable name: leaf_position** 

I created the 2 trait variables: 
https://terraref.ncsa.illinois.edu/bety/variables/6000000261 
https://terraref.ncsa.illinois.edu/bety/variables/6000000262 

How do I link leaf_position covariate to these trait variables so that I can upload the data file?  I'll need to know how to do this for some other data files as well.

NewcombMaria, 2017-10-26:
I found a work-around for this data set by creating a new covariate 'internode_position'.  The data-set uploaded successfully using the newly defined covariate.  For future data file uploads, it still would be good to know how to connect an existing covariate to trait variables using the bulk upload.

gsrohde, 2017-10-26:
@NewcombMaria Your "work-around" did indeed allow you to upload the data.  But that doesn't mean you got the upload wizard to do what you wanted it to do.  Instead of creating "internode_position" covariates for each trait row, it instead inserted the "internode_position" values as traits in their own right.  Nothing was added to the covariates table.

The other problem with this workaround is that it seems that you are duplicating variable names.  What I mean is that it is not good to have two different variables in the database that are essentially the same variable under different names.  So unless "internode_position" is a fundamentally different variable than "leaf_position", it would be best to remove "internode_position" and then take the proper steps to ensure that "leaf_position" will be treated as a covariate of both "stalk_diameter_major_axis" and "stalk_diameter_minor_axis".

Let me explain some things about how the Bulk Upload Wizard works vis-à-vis covariates.

In order for a variable-name column heading to be treated as a covariate rather than as a bona fide trait, it must be listed as a covariate in the trait_covariate_associations table.  You can view the associations provided by this table [here](https://terraref.ncsa.illinois.edu/bety/trait_covariate_associations).  As you can see, "leaf_position" is listed as an optional covariate of both "leaf_length" and "leaf_width".  In order for it to be recognized as a covariate of any other trait, we must add the association to the table.  Perhaps unfortunately, there is no Web interface for adding to this table.  If you want a certain variable to be treated as a covariates of certain other variables, you must request that one of us who has direct database access add new rows to the trait_covariate_associations table.

What I suggest doing now is this:

1. Have me delete all the rows you added today.
1. Have me delete the new variable "internode_position" that you added.
1. Have me add trait-covariate associations for `stalk_diameter_major_axis`—`leaf_position` and `stalk_diameter_minor_axis`—`leaf_position`.
1. After I notify you that this has been done, then you should try re-uploading your data.

NewcombMaria, 2017-10-26:
Many thanks @gsrohde for helpful explanation.  It's confusing since when I define a New Variable there's a drop-down list to select Type as either plant trait or covariate (or ecosystem or some other selections).  I selected 'covariate' for 'internode_position'.  The variable name 'internode_position' actually is a bit more accurate compared to 'leaf_position'.  The stalk diameters were measured within internodes (just below the leaf referred to in leaf_position so the numbers are the same for internode and leaf positions).  I used leaf_position for leaf_length and leaf_width which fits.  Given that internode_position is appropriate for stalk diameters, does it make sense for you to add the trait-covariate linkages using the data already uploaded?

NewcombMaria, 2017-10-26:
This is where it's confusing that it's not possible to distinguish between traits and covariates from the variable definition.  The 'Type' distinction doesn't persist? 
[image]

NewcombMaria, 2017-10-26:
@gsrohde for another data-set uploaded today, did 'measurement_height' as a required covariate get associated with trait variable 'stalk_diameter_fixed_height'?  If not, could you add the trait-covariate association for this data-set as well?  Thanks.

gsrohde, 2017-10-27:
@NewcombMaria 

Regarding the _Type_ field of the "new variable" and "edit variable" forms: Whatever you enter here _does_ persist.  If you visit https://terraref.ncsa.illinois.edu/bety/variables?search=internode_position, you will see the Type listed for `internode_position` is `covariate`, just as you set it to be.

This does not mean, however, that the Bulk Upload Wizard pays any attention to this field when deciding what is a plant trait and what is a covariate.  It doesn't.  As far as I know, this field is purely informational.  In fact, I didn't even know or I didn't remember that `covariate` was a possible choice of type for this field.  Perhaps it _should_ have some bearing on how the Bulk Upload Wizard treats a variable—that is a question to take up with @dlebauer.

Again, `measurement_height` was not treated as a covariate when you did the upload: it is not marked as a covariate as you can see for yourself by visiting https://terraref.ncsa.illinois.edu/bety/trait_covariate_associations.

Again, I recommend that you have me delete this upload set and add the correct associations before you try doing the upload again.  Did you upload anything besides these two sets today?  Is it okay if I go ahead and delete them?

NewcombMaria, 2017-10-27:
If it's necessary to delete the data to make the trait-covariate associations, please do so.  I uploaded only these 2 data-sets yesterday the 26th.  Let me know when they are deleted and then I'll re-upload.  Let's keep the covariate name 'internode_position' - it's the most descriptive term.

gsrohde, 2017-10-27:
Okay, @NewcombMaria, I've delete the 1231 trait rows you added yesterday and I added the following _required_ trait-covariate associations:

1. Each `stalk_diameter_major_axis` trait has required covariate `internode_position`.
1. Each `stalk_diameter_minor_axis` trait has required covariate `internode_position`.
1. Each `stalk_diameter_fixed_height` trait has required covariate `measurement_height`.

Again, you can view these associations at https://terraref.ncsa.illinois.edu/bety/trait_covariate_associations.

You should now be able to re-upload your two data sets.  Let me know if you have any problem.

NewcombMaria, 2017-10-27:
Thanks!  @gsrohde  I was able to successfully upload both data sets.  Fyi this season I'm keeping a record of upload days/times by data file for times like this when we need to backtrack.  (https://docs.google.com/spreadsheets/d/18MezEmq3gSQyPUR5fWarPLPaC6_ticjsi3zhaXe9mJg/edit#gid=568715765 )

NewcombMaria, 2017-10-31:
Thanks again Scott.  This can be closed now.

closed 2017-10-31 by NewcombMaria