terraref/reference-data #199: Wheat Season 3 plot and cultivar information

NewcombMaria, 2017-10-31:
Following up on email communications, if we can assemble the following information for Season 3 (durum wheat experiment Dec 2016 - early April 2017) then the sensor-derived data can be added to the processing system at the same time that the sorghum sensor-derived data is re-processed.  I'll update and notify David and Scott as I organize the following files. 

- [ ] plot names and polygons (WGS84 projection) in shapefiles 

- [ ] cultivars information 

- [ ] sites_cultivars associations

NewcombMaria, 2017-10-31:
@dlebauer or @gsrohde the single spreadsheet in the google folder at the directory below has the Wheat Season 3 cultivar information, site names, and cultivar names.  The site and cultivar column headers are highlighted. 
https://drive.google.com/drive/folders/0B1OR5-zeXagneTBjM1RFYVhrYWM  

This should be the info needed for tasks 2 and 3.  We're working on plot polygons and QC to make sure they line up correctly.  I'll update task 1 as soon as possible.

NewcombMaria, 2017-11-01:
@dlebauer for Task 1, I added 5 files to create shapefiles for plot bounding boxes (whole plots or 2-row plots) to the Google folder at https://drive.google.com/drive/folders/0B1OR5-zeXagneTBjM1RFYVhrYWM 
The 5 files have extensions .dbf, .prj, .shp, .shx.  
I cannot revise the attribute table, or even see what revisions are needed, since I don't have ability to view it in GIS or QGIS.  Would you be able to work with these files to get to the correct site associations for each plot polygon?  They could be merged to the site-cultivar spreadsheet using the Range and Column info.  Pass is synonymous with Column.  Let me know if you need additional info.

NewcombMaria, 2017-11-01:
We also have shapefiles for single-row subplots if that would be useful.  I hope these polygons that I shared are in the correct coordinate system. I think so, but let me know if not.

dlebauer, 2017-11-06:
* Added Wheat to species table ( @NewcombMaria please confirm this is Triticum aestivum)  https://gist.github.com/dlebauer/1d4b5baf94c944894f465a50dc0b5a80#file-wheat_species-sql
* Added cultivars to cultivars table https://docs.google.com/spreadsheets/d/1ijnQlRGV5x45k7m9Eqsc3LitG_jSy-SyWJcr3Q6qu7g/edit#gid=1561416617
* Site information
    * note that I believe these were in UTM (srid 32612) and not ESPG 4386; I converted them.)
* Added sites_cultivars https://docs.google.com/spreadsheets/d/1ijnQlRGV5x45k7m9Eqsc3LitG_jSy-SyWJcr3Q6qu7g/edit#gid=1490335158
* added experiments records to distinguish Border, Uniformity Block, North, and South https://gist.github.com/dlebauer/1d4b5baf94c944894f465a50dc0b5a80#file-new_experiments
* Experiments sites https://docs.google.com/spreadsheets/d/1ijnQlRGV5x45k7m9Eqsc3LitG_jSy-SyWJcr3Q6qu7g/edit#gid=2107185985

All queries saved in https://gist.github.com/dlebauer/1d4b5baf94c944894f465a50dc0b5a80

NewcombMaria, 2017-11-06:
@dlebauer the species name most commonly used for durum wheat is Triticum turgidum var. durum, or  Triticum turgidum ssp. durum.  However, in ITIS I'm seeing that the accepted name might be **Triticum durum**.  See https://www.itis.gov/servlet/SingleRpt/SingleRpt?search_topic=TSN&search_value=798131#null 

If ITIS is the project standard for species name, I suppose it is correct to go with Triticum durum.  In most publications durum wheat is referred to as Triticum turgidum (see for example doi:10.1093/jxb/erw039  Maccaferri et al. 2016. Journal of Experimental Botany 67: 1161-1178.

dlebauer, 2017-11-06:
I've updated the species to _Triticum durum_. The species table from BETYdb is based on plants.usda.gov, which doesn't contain _T. turgidum (var or spp) durum_. I'd go with ITIS over how things are actually published since ITIS is trying to straighten out taxonomies and publications are focused on biology. 

I've also included the combined as well as single row plot definitions.

NewcombMaria, 2017-11-06:
Thanks very much @dlebauer.   Agreed Triticum durum is best for betydb.  Thanks for working on the durum experiment info, and let me know if you need anything else for season 3 metadata.

closed 2018-12-13 by dlebauer