terraref/reference-data #200: Hand-held device data with multiple subsample measurements within plots

NewcombMaria, 2017-11-03:
It's not clear to me how to indicate that subsample measurements within a plot (taken on random plants in the same plot) are associated/connected with each other.  This issue relates to a discussion on the use of entity in issue #165 and a dataset referred to in closed issue #162 (this was an example of an incorrect use of entity since the sample 1, sample 2, etc were on random plants).  
Subsample data points were recorded typically when measurements were collected using hand-held devices that targeted a very small area, and multiple subsamples (5 or 6) more accurately captured the variability within the plot.  There are two different cases:

1) the hand-held device recorded a time-stamp with every measurement, in which case the specific time-stamp makes the measurement unique within the plot (for example stomatal conductance using a porometer)
2) in other cases the exact time is not known for each measurement, but an approximate time is known for when all 6 subsample measurements were made for each plot (for example leaf temperature using an Apogee hand-held infrared radiometer) 

@dlebauer and/or @gsrohde For both cases, how do I correctly associate all 5 or 6 subsamples with each other to represent one plot or site?

dlebauer, 2017-11-03:
@NewcombMaria you are correct that entity should not be used to track random samples within a plot. There are two options:

1. Compute plot level stats (mean, SE, n) and enter those
2. Enter all six samples per plot without uniquely identifying them as an entity. That the measurements were taken at the same time and in the same plot will be captured by the plot(sitename) and datetime fields. 

The first option (computing plot level stats) is essentially what we are doing with the tractor based sensors - averaging a bunch of points within the plot at each time point (the scanner sensors are an extreme case). Ultimately, this is what a breeder will be most interested in, but it leaves out potentially useful information not captured by the mean and SE ... illustrated nicely by this figure from [Weissburger et al 2015](http://journals.plos.org/plosbiology/article?id=10.1371/journal.pbio.1002128)

[image]

My recommendation is to upload all of the individual points. If the exact time is not known, you can use the same approximate time to all six points and optionally add a value of timeloc (this field has rarely been used and is not supported by the bulk upload; I can update these in the database if you want)

These are values for timeloc ([see documentation](https://pecan.gitbooks.io/betydbdoc-dataentry/content/Adding%20Traits%20and%20Yields.html#timeloc)); fractions can be used to indicate intermediate values:

Timeloc | Definition
-- | --
9 | no data
4 | time of day i.e. morning, afternoon
3 | hour
2 | minute
1 | second

You could also assume measurements were measured at a constant pace within each plot and interpolate but it is not clear if this will be worth the effort.

NewcombMaria, 2017-11-03:
Thanks @dlebauer .  Makes sense, the timepoints will connect them.  I'll keep the individual data points and when exact time is unknown will give the same approx time for all subsamples with a level of confidence (timeloc).

gsrohde, 2017-11-09:
@NewcombMaria, @dlebauer The trait-upload API allows setting the `timeloc` value only indirectly: it will be set to 9 if a datetime of the form `2017-11-09` is given and will be set to 1 if a value of the form `2017-11-09T13:47:13` is given.

If all the timeloc values should be the same, it shouldn't be too hard to adjust this after the fact.  For example, if all measurements are to the nearest hour, leave the minute and second portions of the datetime as `00`, note the ids of the added traits, and then do a direct database update to set the timeloc values to 3.

NewcombMaria, 2017-11-10:
Thanks @gsrohde.  Most of the time data will be accurate to within a minute.  The porometer, Flir spot camera, Multispeq and Dualex instruments all record the time.  There are only 2 datasets that I recorded the approximate time manually, which might be plus or minus about 3 minutes.  If the timeloc can't be set for the approximate times it could be described in the trait variable definition.

NewcombMaria, 2017-11-16:
The API upload process with local_datetime data seemed to go well for 4 datasets so far.  This can be closed now.

closed 2017-11-16 by NewcombMaria