terraref/reference-data #201: Linking trait data in Betydb to associated images??

NewcombMaria, 2017-11-10:
@dlebauer and @gsrohde I'm working on a dataset for upload to BETYdb that has numerical leaf temperature data values along with an associated image (.bmp) for each temperature value.  The data were collected using a FLIR spot camera.  I've extracted the temp value shown in the upper left of each image, see example below.  I've included the image file name that corresponds to the leaf temperature in the spreadsheet.  (I don't have the spreadsheet completed yet, but will have it ready sometime tomorrow.)

Is there a way to link the images to the trait variables?  Both Geoff Morris and Wasit Wulamu were interested in having access to the images along with the leaf temperature trait data.

MAC Field Scanner   Season 4 Range 40 Column 15 W     2017-06-26   14:11   FLIR00120.bmp     32.0  

[image]

dlebauer, 2017-11-10:
There is, but we should revisit this during our 2018 planning. But in the short term, if the image names currently have the plot name, date, and time, we can put them into a folder like sites/ua-mac/Level1/FLIR_handheld. It will then be straightforward to query them and upload them into Clowder later.

NewcombMaria, 2017-11-10:
Thanks.  Works for me.  I'll update here when I have the folders organized.

NewcombMaria, 2017-11-14:
The images are in folders by date (7 dates), along with the spreadsheet for the FlirTG1167 data in the following directory: 
https://drive.google.com/drive/folders/1TG5_qHqssuAzzAPBbvTC7DnEvxCIxFJY

dlebauer, 2017-11-14:
@NewcombMaria picking from https://github.com/terraref/reference-data/issues/204#issuecomment-344364451 here where the focus is on the handheld FLIR data ... 

the entity should be something uniquely identifying - its not clear why there are duplicate image_ids on different dates (e.g. 6 of FLIR00001.bmp) but if they are just sequential ids that restart each day (?) then I would suggest using date + image_id as a unique identifier. (plot + image_id would not be unique). But perhaps it resets every time it gets to 99, then using datetime + image_id or date + plot + image_id would be appropriate for entity. 

Then in notes, if the purpose is to help someone find the file, then it would be useful to provide the full path (like sites/ua-mac/Level1/FLIR_handheld/YYYY-MM-DD/FLIR00001.bmp) so that it is easy to look up the file.

NewcombMaria, 2017-11-14:
@dlebauer the image numbers reset every time I downloaded them.  Most of the time I downloaded after every data collection day, but sometimes not and the numbering continued until the next download.  date+image_id will be unique.  

Your suggestion for notes makes perfect sense.  Can I use 'sites/ua-mac/Level1/FLIR_handheld' with the assumption we'll transfer the images in the near future?  Or is there a better location?

dlebauer, 2017-11-14:
yep, you can use sites/ua-mac/Level1/FLIR_handheld

NewcombMaria, 2017-12-05:
@dlebauer have the FLIR_TG167 images been transferred yet to 'sites/ua-mac/Level1/FLIR_handheld' from the google folder at https://drive.google.com/drive/folders/1TG5_qHqssuAzzAPBbvTC7DnEvxCIxFJY 

If not, do I need a user-id and password for cache server access to do the transfer?  If they have been transferred, can this be closed?

dlebauer, 2017-12-05:
@jdmaloney what does Maria need to do to transfer these files to the cache server?

Paheding, 2017-12-06:
Per this issue, @NewcombMaria @dlebauer If we could link all ground measured trait data (e.g., Plant Count, biomass, yield, etc ) in Betydb to associated images (RGB, hyperspectral, etc.) in Globus will be extremely helpful from the perspectives of plant science and remote sensing. 

I have gotten very interesting trait data provided by @NewcombMaria, and I really want to find the associated images but it seems that it is not very convenient to do so. One suggestion is that add trait data to metadata of the associated images.

dlebauer, 2017-12-06:
@Paheding the lookup you mention is intended to be done based on geospatial and temporal information.

I think there is a basic overview here: https://github.com/terraref/tutorials/blob/master/sensors/06-list-datasets-by-plot.md

@jterstriep have you had a chance to document this in more detail? Is the API up and running yet?

dlebauer, 2017-12-06:
@Paheding one caveat is that for these handheld FLIR files I don't think that there is any geospatial metadata. So the spatial+temporal query approach will not work for these files unless we add the geospatial metadata.

dlebauer, 2017-12-07:
@jterstriep will do the following:

* submit a PR to terrautils w/ a function defined in https://github.com/terraref/documentation/issues/157
   * assign to @Paheding and @ZongyangLi for review
* deploy the webservice 
* update the documentation

Paheding, 2017-12-07:
@dlebauer Sounds a good plan. I will review it once it'd done.

Paheding, 2018-01-19:
This issue can be referred to [https://github.com/terraref/reference-data/issues/219](url)

dlebauer, 2018-01-25:
* @jterstriep has the docker container, will update documentation
* @craig-willis will deploy to workbench kubernetes

craig-willis, 2018-02-05:
@jterstriep I have this running on https://sensorquery.workbench.terraref.org.  I had to make one change to get the image to build and another because Workbench has different paths mapped for the TERRAREF_BASE.  Let me know what you think and if it's working as expected:

PR
https://github.com/terraref/service-products/pull/1

max-zilla, 2018-02-21:
@craig-willis need a status update.

craig-willis, 2018-02-22:
PR is still open, waiting for feedback from @jterstriep

max-zilla, 2018-03-14:
@jterstriep closed the PR but incorporated changes into his own. 

https://github.com/terraref/service-products/pull/2

dlebauer, 2018-03-14:
Note that the web services that have been the topic of discussion here are not directly relevant here. The case here is that Maria has collected images in the field.

Next steps: @NewcombMaria where are these images so we can take a look at how to support this use case?
Do you have a table of image names and locations (or plot names)? If so, we can use this information to 1) organize the images by date and plot name and 2) link these in the database so the above works.

NewcombMaria, 2018-03-15:
@dlebauer the handheld FlirTG1167 images are all within a google drive folder, copied into separate folders by date (7 dates), along with the spreadsheet for the FlirTG1167 data that already has been uploaded to BETYdb. 
https://drive.google.com/drive/folders/1TG5_qHqssuAzzAPBbvTC7DnEvxCIxFJY 

The spreadsheet is also there:    Sorghum season4_StomatalRegulation_FlirTG167_7dates_8plots.xlsx 
In the spreadsheet the image info is listed as an entity, and the notes column has the directory info, something like: 'see image file: sites/ua-mac/Level1/FLIR_handheld/2017-06-14/FLIR00001.bmp'

dlebauer, 2018-03-15:
Use case - two week stomatal conductance campaign, taken alongside porometer measurements. 

next steps:

* [ ] move image files to appropriate folders on the file system
   * put in raw_data/FLIR_handheld/YYYY-MM-DD/plotname`
   * update file path in entities.notes

dlebauer, 2018-03-15:
Maria - to clarify, in the [Season4 Stomatal Regulation Campaign](https://drive.google.com/open?id=1TG5_qHqssuAzzAPBbvTC7DnEvxCIxFJY) folder there are two spreadsheets with leaf temperature data:

* one with FLIRTG167 in the name: [Sorghum season4_StomatalRegulation_FlirTG167_7dates_8plots](https://drive.google.com/open?id=19VbYuQkhaAQvSsjSc-RVlxijm0cbgp1wzhg5Q9x0PFo) 
* the other with ApogeeIRT in the name: [Sorghum season4_StomatalRegulation_ApogeeIRT_7dates_8plots](https://drive.google.com/open?id=1J2Wp3gQ38H6bdSFZXGiu2T3Cmh0FHUJEoQ-pKrwrEhU)

However, these both have the same method ('Leaf surface temperature and stomatal conductance estimates using hand-held instruments') and the same trait ('surface_temperature_leaf") name. We need to use the methods field to differentiate these two approaches to measuring the leaf temperature. But if we are measuring the canopy temperature (as I suspect the Apogee does), then we should call it surface_temperature_canopy or similar. 

Also, please confirm that these are duplicates of the above files and can be deleted
* [Sorghum season4_StomatalRegulation_FlirTG167_7dates_8plots](https://drive.google.com/open?id=1BErpYk0PXFxKOt5PqkusrsmUkOnJw0iC) in the FLIR_TG167_Images_2017-06-23 sub folder
* [Sorghum season4_StomatalRegulation_ApogeeIRT_7dates_8plots](https://drive.google.com/open?id=1DSY3hQTmVAGMxBRi5O1OkcBpG4W8bOl9) in FLIR_TG167_Images_2017-06-14 sub folder

dlebauer, 2018-03-15:
I did the following, which I believe will resolve this issue:

* put these images in globus under sites/ua-mac/raw_data/FLIR_TG167_handheld/YYYY-MM-DD
* updated paths from ea. trait record to image file (see [update queries](https://gist.github.com/dlebauer/4578a4478f95ca2dff3c174069ef3c92#file-update_flir_trait_notes-sql) and [how they were concatenated](https://docs.google.com/spreadsheets/d/1-fcwhxObdqXxcSBPYxwLLeq84W7PZof-GFpUmabfTWU/edit#gid=0))
* added this [metadata.csv](https://gist.github.com/dlebauer/4578a4478f95ca2dff3c174069ef3c92#file-metadata-csv) file to the FLIR_TG167_handheld directory 

@NewcombMaria if this is suitable, please close this issue and open a new one if we need to address the methods issue mentioned above in https://github.com/terraref/reference-data/issues/201#issuecomment-373468463

NewcombMaria, 2018-03-15:
Thanks @dlebauer.  The thinking behind using the same method description ("Leaf surface temperature and stomatal conductance estimates using hand-held instruments') is that we deliberately followed the identical method using both instruments to be able to have independent instruments collecting data for estimates of leaf temp for cross-comparisons.  The handheld IRT model that was used (MI-230) has an 'Ultra-Narrow Field of View' and truly was measuring temperature of a single leaf - we aimed at a single sun-lit leaf.  The trait 'surface_temperature_leaf' is the most accurate in this case because of the ultra-narrow FOV.  

Confirmed that the 2 files in the dated sub-folders are duplicates and I removed them both.

NewcombMaria, 2018-03-15:
@ the metadatafile.csv file looks great.  Thanks for transferring the images.  This issue can be closed.

closed 2018-03-15 by NewcombMaria

dlebauer, 2018-03-15:
@NewcombMaria 

Thanks for the explanation. Sorry that this is so obtuse, but I think this is a nice illustration of how the data is intended to be structured.

In this case, the methods field should be used to distinguish among the two instruments and the entity would be used to link measurements made on the same leaf (so the name could be named "replicate leaf 1 in plot xyz" or something along those lines. We did something similar in the past when you labeled plants by color and returned to them over time.

I'll open a new issue

dlebauer, 2018-03-15:
@NewcombMaria see https://github.com/terraref/reference-data/issues/237