terraref/reference-data #204: Adding trait-covariate associations before uploading trait data to BETYdb

NewcombMaria, 2017-11-14:
Datasets with newly defined trait variables that need to be associated with covariates are ready for upload to BETYdb.  The associations that need to be set up in the trait_covariate_associations table by either @gsrohde or @dlebauer are defined below.  These datasets have local_datetime data which requires the API upload process, which I'll be attempting to learn today.  

- [ ] trait:  **leaf_stomatal_conductance** (https://terraref.ncsa.illinois.edu/bety/variables/6000000229) 

- needs to be associated with 3 newly defined covariates (optional):  **sensor_calibration** (https://terraref.ncsa.illinois.edu/bety/variables/6000000274), **leaf_sensor_RH** (https://terraref.ncsa.illinois.edu/bety/variables/6000000275), and **filter_sensor_RH** (https://terraref.ncsa.illinois.edu/bety/variables/6000000276) 

- leaf_stomatal_conductance has an existing optional covariate (ambient_air_temperature), and the new dataset will also have **ambient_air_temperature** (https://terraref.ncsa.illinois.edu/bety/variables/6000000230) as a covariate 

- [ ] 4 newly defined trait variables that are grouped in one dataset all need to be associated to one covariate (optional).  The new trait variables are **chlorophyll_index** (https://terraref.ncsa.illinois.edu/bety/variables/6000000280), **flavonol_index** (https://terraref.ncsa.illinois.edu/bety/variables/6000000281), **anthocyanin_index** (https://terraref.ncsa.illinois.edu/bety/variables/6000000282), and **NBI_nitrogen_balance_index** (https://terraref.ncsa.illinois.edu/bety/variables/6000000283).  The covariate that all 4 need to be associated with is named **sensor_emitter_temperature** (https://terraref.ncsa.illinois.edu/bety/variables/6000000279).  

- [ ] newly defined trait variable **surface_temperature_leaf** (https://terraref.ncsa.illinois.edu/bety/variables/6000000272) needs to be associated with 1 newly defined covariate variable: i**mage_id** (https://terraref.ncsa.illinois.edu/bety/variables/6000000273).  Linking images with FLIR thermal spot camera data is discussed in issue #201

gsrohde, 2017-11-14:
@NewcombMaria Should the `image_id` covariate be optional like all the other covariates you mention are?  Also, I didn't see any mention in issue #201 of this method of linking images to traits.  And if the image id is to be stored in `covariates.level`, it will have to be numeric.

dlebauer, 2017-11-14:
image_id sounds like it should be an entity rather than a covariate

gsrohde, 2017-11-14:
I think that's what we were thinking when this issue first came up early on.  The only potential problem I see is if a trait associated with an image might more properly be associated with a more narrowly-defined entity: there can be only one entity per trait.

gsrohde, 2017-11-14:
Assuming we won't use `image_id` as a covariate, this is the insert statement for adding the associations listed above:
```
INSERT INTO trait_covariate_associations VALUES
       (6000000229, 6000000274, 'f'),
       (6000000229, 6000000275, 'f'),
       (6000000229, 6000000276, 'f'),

       (6000000280, 6000000279, 'f'),
       (6000000281, 6000000279, 'f'),
       (6000000282, 6000000279, 'f'),
       (6000000283, 6000000279, 'f');
```
@dlebauer Shall I go ahead and run this?

dlebauer, 2017-11-14:
@gsrohde yes please run that

for the current use case, I think that a single entity would work. 

In the future if there is a need for > 1 entity per trait we can either use the parent-child relationships to map many entities to one trait or implement and use the generic table https://github.com/PecanProject/bety/issues/505

gsrohde, 2017-11-14:
Done.  Can this be closed?

NewcombMaria, 2017-11-14:
Before closing, I want to be certain I understand the entity designation formatting for the images.  The spreadsheet with the data is called 'Sorghum season4_StomatalRegulation_FlirTG167_7dates_8plots.xlsx' at https://drive.google.com/drive/folders/1TG5_qHqssuAzzAPBbvTC7DnEvxCIxFJY
The complete name for the image is listed in the 'notes' column (col S), and the numerical component is listed in the 'image_id' column (col T).  How should I identify the entity?  I can concatenate the site name with the image name, for example '**MAC Field Scanner Season 4 Range 9 Column 3 W FLIR00001.bmp**'?  Or can the entity simply by the image name 'F**LIR00001.bmp**'?

dlebauer, 2017-11-14:
@NewcombMaria I've answered your question from https://github.com/terraref/reference-data/issues/204#issuecomment-344364451 in issue https://github.com/terraref/reference-data/issues/201#issuecomment-344384389, but to summarize, for entity use datetime + image_id and for notes use the full directory path and file name

NewcombMaria, 2017-11-15:
Ok, I created unique identifiers as date + image and designated these as an entity instead of covariate.  This issue can be closed now.

closed 2017-11-15 by NewcombMaria