terraref/reference-data #208: Trait-covariate associations need to be defined for Multispeq field measurements dataset

NewcombMaria, 2017-11-16:
@gsrohde and @dlebauer all of 35 trait variables and 6 covariates are now defined for the photosynthesis-related parameters in the Multispeq field dataset.  All of the trait variables are new, and 5 of the 6 covariates are new.  My thoughts are that the covariates probably should be associated with all of the traits in the dataset, but it's not obvious how these covariates should correspond to all of the traits.  The spreadsheet is named 'Sorghum season4_Multispeq_Aug_StressPhotosynthesis_final.xlsx' in the folder directory https://drive.google.com/drive/folders/1MVkQpJ8f_gCHZP17j63tXaAGCT1R5tAO 

The 6 covariates are highlighted in orange in the headers:  (blue header highlights are columns not for upload)

- ambient_humidity (https://terraref.ncsa.illinois.edu/bety/variables/6000000317)

- ambient_air_temperature (https://terraref.ncsa.illinois.edu/bety/variables/6000000230)

- light_intensity_PAR (https://terraref.ncsa.illinois.edu/bety/variables/6000000318)

- leaf_angle_clamp_position (https://terraref.ncsa.illinois.edu/bety/variables/6000000319)

- pitch (https://terraref.ncsa.illinois.edu/bety/variables/6000000320) 

- roll (https://terraref.ncsa.illinois.edu/bety/variables/6000000321) 
Is it possible/reasonable to make these trait-covariate associations??  Do you need more info from me?

dlebauer, 2017-11-16:
Maria. First I have to say that this is dataset is amazing, and along with all of the other data you've collected, curated, and contributed deserves more kudos than fit here. 

Back to the question... I'm inclined to make these covariates as traits, but will all a few points of clarification 
1. Were these were taken while the sensor was clipped to the sensor?
1. do pitch and roll  account for the native state of the leaf before clipping on the sensor? Or are these artifacts of the sensor?

Fort the environmental measurements ... something like temperature and humidity during measurement reflects the biological processes ir transportation. These should not be confused with our air_temperature variable that reflects the well mixed atmosphere on our met station.

gsrohde, 2017-11-16:
@NewcombMaria I think you missed defining variable `contactless_temp_C`.

Once that is defined, the statement
```
INSERT INTO trait_covariate_associations
    SELECT t.id, c.id, 'f' FROM
            (SELECT id FROM variables WHERE name IN (
                                                     'absorbance_420',
                                                     'absorbance_530',
                                                     'absorbance_605',
                                                     'absorbance_650',
                                                     'absorbance_730',
                                                     'absorbance_850',
                                                     'absorbance_880',
                                                     'absorbance_940',
                                                     'contactless_temp_C',
                                                     'ECSt',
                                                     'FmPrime',
                                                     'FoPrime',
                                                     'Fs',
                                                     'FvP/FmP',
                                                     'gH+',
                                                     'leaf_temperature_differential',
                                                     'leaf_thickness',
                                                     'LEF',
                                                     'NPQt',
                                                     'Phi2',
                                                     'PhiNO',
                                                     'PhiNPQ',
                                                     'qL',
                                                     'qP',
                                                     'relative_chlorophyll',
                                                     'RFd',
                                                     'SPAD_420',
                                                     'SPAD_530',
                                                     'SPAD_605',
                                                     'SPAD_650',
                                                     'SPAD_730',
                                                     'SPAD_850',
                                                     'SPAD_880',
                                                     'surface_temperature_leaf',
                                                     'vH+')) AS t
        CROSS JOIN
            (SELECT id FROM variables WHERE name IN (
                                                     'ambient_humidity',
                                                     'ambient_air_temperature',
                                                     'light_intensity_PAR',
                                                     'leaf_angle_clamp_position',
                                                     'pitch',
                                                     'roll')) AS c;
```
should insert the 210 new rows in the `trait_covariate_associations` table.

@dlebauer Note that each CSV file row will generate 35 new traits and 210 new covariates.  The 210 covariates will be in 6 groups of 35 which will differ only in their `trait_id`.  At least covariate rows are relatively small (9 columns).

gsrohde, 2017-11-16:
@NewcombMaria @dlebauer If we make the covariates traits, then we'll just get 41 traits and 0 covariates per row.

dlebauer, 2017-11-16:
Sorry I posted too quickly I'm reading the method paper http://rsos.royalsocietypublishing.org/content/3/10/160592

Maybe all but rh and T should be covariates. (Is your measurement of par taken horizontally above the canopy like the weather station?)

To answer you original ? I think all recognized covariates will be associated with all recognized traits within a row. In this case I think that means there will be 35x3 or 35x6 covariates for each row added. 

(as an aside, this may not seem optimal. Indeed a data first principle suggests each data point have a single canonical representation and a database first principle called normalization suggests better ways to structure the database so that each covariate values wouldn't be repeated. But what we have works. And in many cases across the project we are duplicating data so it is easier to use / scale / find what you need. The planned generic table will provide more backend options but the import and export formats / API will stay the same.

gsrohde, 2017-11-16:
>  I think all recognized covariates will be associated with all recognized traits within a row.

David, this is only true if there is a row in `trait_covariate_associations` associating each recognized trait with each recognized covariate.

To summarize: A variable used as a column heading will be "recognized" if it matches the name of a variable in some row of the variables table.  It will be recognized as a covariate if it is listed as a covariate of some trait in the `trait_covariate_associations` table; but it will be treated as a covariate only of those traits it is explicitly associated with in that table.  Any recognized variable not listed as a covariate of some other variable will be treated as a trait variable.

NewcombMaria, 2017-11-16:
Thanks @gsrohde and @dlebauer .  Scott, I made a mistake with the header 'contactless_temp_C' which should be ignored and not uploaded.  It's another term for surface temperature of the leaf, which is already defined as 'surface_temperature_leaf' in units K.  I fixed the shreadsheet by adding _ignore and highlighting in blue.  

David, the question about PAR:  the handheld measurement is a bit different from the horizontal measurement by the weather station because it's measured at the perpendicular from the instrument clamp that goes on the leaf and often the instrument was at an angle different from horizontal.  

Same for the other measurements like ambient temp, ambient humidity, pitch and roll.  These are measured while the sensor was clamped on a leaf.  All are a function of where/how the sensor is placed on the leaf.  Many of the leaves were bowed or curved convex, and the pitch, roll, PAR, and even air temp and humidity were influenced by where in the curve and where in the canopy the sensor was placed on the leaf.  Still, it's not obvious which to consider covariates.  I don't think they quite fit as plant traits, but some are close fits for traits.

dlebauer, 2017-11-16:
@NewcombMaria it sounds like all of these are influenced by the plant. Although it is a grey area I believe they should be traits. 

By contrast, when leafT (or PAR or humidity) are recorded as a covariate for parameters derived from LICOR 6400 photosynthesis measurements, the leafT covariate doesn't reflect the leaf's natural environment because the leaf temperature is actively controlled by the sensor. In this case, it sounds like the instrument is measuring rather than controlling how the leaf exists.

We can always revisit later ...

NewcombMaria, 2017-11-16:
Defining them as traits works for me.  It's correct that the instrument is only clamped on the leaf for about 20 seconds to measure leaf status (not change the status) over a small area of the leaf.  When I define the variables, there's a drop-down list to select either 'plant trait' or 'covariate'.  I selected covariate for these 6.  Scott has suggested that it doesn't make much difference which is selected from the web interface variable definition.  Does anything need to be changed?  If not, I can upload with the variables defined as they are now.

dlebauer, 2017-11-16:
> Scott has suggested that it doesn't make much difference which is selected from the web interface variable definition. 

That is correct ... it doesn't affect how the data are stored - the traits_covariates_associations table does that.

NewcombMaria, 2017-11-16:
Thanks.  I'll upload as trait variables.

NewcombMaria, 2017-11-16:
Because ambient_air_temperature was already designated a covariate from an earlier dataset, I get the error message: "errors": "Covariate variable(s) (ambient_air_temperature) with no matching trait variable was found in the CSV file heading"

The easiest way to to make it a trait would be do change the name and define it as something like 'ambient_air_temperature_multispeq' or 'proximal_air_temperature'.

NewcombMaria, 2017-11-16:
I changed 'ambient_air_temperature_ to a trait named 'proximal_air_temperature'.  (It seems to fit, proximal to the plant canopy and makes sense for a trait).  There's another problem with the data file.  I get the endless error...  "errors": "{:schema_validation_errors=>[\"Element 'trait', attribute 'mean': '' is not a valid value of the atomic type 'xs:double'.\", \"Element 'trait', attribute 'mean': '' is not a valid value of the atomic type 'xs:double'.\", \"Element 'trait', attribute 'mean': '' is not a valid value of the atomic type 'xs:double'.\", 

I see 2 potential problems, but I'm guessing and it could be something else causing the error. 
1) some of the values for vH+ are in the format '2.95E-05'
2) some of the trait names have symbols, for example vH+

Any suggestions?

dlebauer, 2017-11-16:
Try fixing 1 first then 2?

NewcombMaria, 2017-11-16:
I just tried fixing the exponent formatting and still get the error message.  I'm going to let it go until tomorrow.  Maybe @gsrohde will see an obvious problem.

NewcombMaria, 2017-11-16:
The general outline in the documentation https://pecan.gitbooks.io/betydbdoc-dataentry/content/trait_insertion_api.html has a statement:
"A trait element must include a mean attribute with a value of type double. An exponent may be used; for example, in place of "0.0123" one may write "1.23E-2"." 

The error message mentions attribute 'mean'.  Not sure what it means that it must include a mean attribute.  I'll try changing the trait variable names that have mathematical symbols (+ and /).

NewcombMaria, 2017-11-16:
Making the changes below to trait variable names still resulted in the same error message (errors": "{:schema_validation_errors=>["Element 'trait', attribute 'mean': '' is not a valid value of the atomic type 'xs:double'.", "Element 'trait', attribute 'mean': '' is not a valid value of the atomic type 'xs:double'.",)

gH+ changed to gH_plus;  vH+ changed to vH_plus; FvP/FmP changed to FvP_FmP_ratio

@gsrohde or @dlebauer when you have a chance I'll need your help sorting out the cause of the error.  If the symbols in the variable names is not the problem, I'd like to change back to gH+, vH+, and FvP/FmP.

gsrohde, 2017-11-16:
Symbols in the name shouldn't be a problem.  If I have time, I'll take a look to see what the problem might be.

NewcombMaria, 2017-11-16:
@gsrohde is there a limit set on the numbers of trait variables that can be uploaded from one spreadsheet, or a limit to the number of columns?  The .csv file has 64 column, and 40 variables.

gsrohde, 2017-11-16:
@NewcombMaria There is no hard limit—the only problem would be if the server times out.

In any case, this wouldn't cause the validation errors.  These are probably because you have some blank cells in some of the variable columns.  I thought I had fixed things so that these would be allowed (at least for trait variables if not for covariate variables), but maybe I did that only for the bulk upload.  I'll check.

NewcombMaria, 2017-11-16:
@gsrohde there are 8 variable columns that have empty cells.  If you think that's the most likely source of the error I can split those 8 variables into a separate file and test the upload with the majority of variables that have values in all cells.  The 8 include ECSt, gH+ (temporarily gH_plus), vH+ (temporarily vH_plus), absorbance_420, absorbance_530, absorbance_605, absorbance_650, and SPAD_420. 

I'd like to change the variable names with symbols back to the original, but I'll wait until we have this sorted out so I don't add complexity.

gsrohde, 2017-11-16:
There are some problems with this approach.  I have another idea, but I want to test whether I think it's feasible.  Then I'll get back to you.

NewcombMaria, 2017-11-16:
Thanks @gsrohde for your help!

gsrohde, 2017-11-16:
@NewcombMaria 

Every cell in every column corresponding to trait variable _must_ contain a valid number.  In particular, it can't be blank, and it can't be "null" or "NA" or "NaN".

In the previous set, where it was only one column where there might or might not be a value, it wasn't too difficult to split the file into the two cases—one file where there was a value and one where there was not.  But with eight columns, there are potentially 256 (2^8) cases.

Unless you are content just to omit the data from columns that aren't always populated with a valid number, I suggest the following work-around:

1. Replace the content of every cell that is blank or contains "null" with a valid-but-unlikely-to-occur number, let's say "-9999999".
2. Do the upload with the altered file.
3. Delete all traits just uploaded having "-9999999" in the "mean" column.  ("mean" is the name of the trait column that stores the value of the measured variable.  Presumably this was named with the case of a sample size larger than one in mind.)

The only other feasible alternative I see is this:
1. Delete (or ignore by renaming) all columns where any cell in the column is blank or not a number.
2. Upload the altered file.
3. Change the API to allow blank cells—to just not make traits for those cells.
4. Delete the rows from the original upload.
5. Re-upload the original file (without any columns deleted).

I'm testing the altered file now on my own BETYdb instance just to be sure there aren't other problems.  It's taking a while, so I hope we don't need to split up the file for this reason.

NewcombMaria, 2017-11-16:
Thanks @gsrohde .  Whichever approach you think best is good with me.  Entering -9999999 for empty cells and deleting after uploading seems reasonable.  Let me know after your test.  I've been splitting some of the larger files into subsets of rows to make the upload go faster without timing out.  

You will need to instruct me how to delete uploaded traits with -9999999 in the 'mean' column.

gsrohde, 2017-11-16:
@NewcombMaria I think the first approach is reasonable.  I will have to delete the rows since you don't have direct database access.

My test has been running for over 75 minutes.  Even though there are 1780 data rows with 40 traits per row, this seems like a long time.  I don't think my server has timed out, but the terra-ref one might.

If you are going to split files, I recommend replacing the blanks with -9999999 before splitting.  It's easier to do a search-and-replace on one file than on several.  With a text editor, I did a search for ",," (representing an empty cell) and replaced with ",-9999999,".  There were also three "null" values to replace, so I also did a search for "null" and replaced with "-9999999".  It remains to be seen if there are other non-numbers that need to be replaced.

Are there other large files that you have yet to upload besides the one mentioned at the top of this issue?  You said "I've been splitting some of the larger files", which implies more than one.  If so, are they as large as this one?

I was thinking of writing a script to split files automatically, keeping the header line in each resulting file.

gsrohde, 2017-11-16:
My upload to my test server just finished.  It took about 90 minutes.

NewcombMaria, 2017-11-17:
Great.  Thanks @gsrohde .  I'll replace empty cells and "null" values with -9999999 before splitting to smaller files and then upload the entire dataset.  This file that we're working on is the last and largest of the large files..............at least for this 2017 season.  I'll let you know tomorrow if the upload is successful.  If it goes well I'll ask you to delete the traits with -9999999.  

A script to split files would be useful to reduce chances of duplicating or leaving out rows by mistake...........for future datasets.

NewcombMaria, 2017-11-17:
Success with the upload!  At least as far as I can tell.  I encountered a curl message for one of the sub-files while it was listing the trait id numbers '(56) send failure', but the trait id number totals fit with the correct trait counts.  When I search on several of the trait variables in BETYdb, it's consistently showing that there are 1780 trait values for each.  It looks like all the data uploaded successfully. 

The next step is to delete the traits that have '-9999999' in the mean column.  @gsrohde let me know if anything looks odd when you have a chance to look at the trait values and delete the ones that were originally null or NA etc.  Thanks!

gsrohde, 2017-11-17:
@NewcombMaria Okay, I deleted the traits that had `-9999999` in the `mean` column.  Can this issue be closed?

NewcombMaria, 2017-11-17:
Thanks!!  It can be closed.

closed 2017-11-17 by NewcombMaria