terraref/reference-data #213: Download scanner data

ebabaeian, 2017-12-18:
Hello,
I am going to download a sample VNIR and SWIR image to see the quality and quantity of the scanner data. I have tried to do this via space through https://terraref.ncsa.illinois.edu/clowder/spaces, as:
1) Maricopa Agricultural Center Field Scanner - 2017 Data Release Level 1
2) Sample Data
but it seems that the data are not downloadable and I see an error page. On the other hand, I don't find any SWIR image in the dataset. Only RGB and VNIR data are exist which are not downloadable. 

Any thoughts/
Thanks very much
Ebrahim Babaeian

  

 

_PLEASE USE THIS TEMPLATE when submitting a new issue_

### Description (REQUIRED)

 
### Completion Criteria

dlebauer, 2017-12-18:
@ebabaeian 

I suspect some of what you are seeing is the most 'recent' files that show up by default in the web interface, but may not be finished processing (datasets are created and then populated). Probably the best place to start is using workbench (workbench.terraref.org) where you can explore the filesystem (e.g. using bash, R, python, or any environment from dockerhub.

* SWIR data products were not included in the Nov 2017 beta release because we do not have the hardware (SWIR point radiometer for downwelling radiation) required for our calibration pipeline (github.com/terraref/extractors-hyperspectral).
* For VNIR, the data and metadata are in the [VNIR Hyperspectral NetCDFs](https://terraref.ncsa.illinois.edu/clowder/collection/5a0cc7e64f0ca87e15ba3163) but the files are so large, it would be worth looking into using the THREDDS server (terraref.ncsa.illinois.edu/thredds/catalog.html), exploring in workbench.terraref.org, or transferring data using Globus. 
* RGB should be downloadable via the clowder interface, e.g. if you go to the [2017 Data Release Level 1 space](https://terraref.ncsa.illinois.edu/clowder/spaces/59d3e9594f0c888ad6ca1279) 
    * currently the [RGB GeoTIFFs - 2016-04-29__14-18-56-444](https://terraref.ncsa.illinois.edu/clowder/datasets/5a37f9e44f0ccf88dc7be73b?space=59d3e9594f0c888ad6ca1279), with a creation date of today (Dec 18) show 0B files. In any case, the download appears to work - can you check again?
    * by contrast the [RGB GeoTIFFs - 2016-06-12__16-40-04-127](https://terraref.ncsa.illinois.edu/clowder/datasets/5a26ba7e4f0cda159545d6a6?space=59d3e9594f0c888ad6ca1279) created on Dec 5 show the file sizes (~25MB) and if you click on one of the files e.g. [rgb_geotiff_L1_ua-mac_2016-06-12__16-40-04-127_right.tif ](https://terraref.ncsa.illinois.edu/clowder/files/5a26ba8b4f0cda159545d812?dataset=5a26ba7e4f0cda159545d6a6&space=59d3e9594f0c888ad6ca1279), you can find the path on disk under 'filename'. This file path should be accessible through the workbench or Globus.

dlebauer, 2018-01-11:
@ebabaeian have you been able to access the VNIR data? Can this issue be closed?

ebabaeian, 2018-01-12:
Hi D. Lebauer
Yes, I am downloading some VNIR and SWIR data for Dec 2017.

Thanks

closed 2018-01-12 by dlebauer