terraref/reference-data #215: StereoTop image manual quality control check

NewcombMaria, 2017-12-31:
With the intention to catch any potential problems early and ensure that camera settings are optimized for downstream extractors I've been viewing a few images from the cache server.  Some images look overexposed to my eye, but I need input from others involved in computational processing and/or image analysis.   I converted images captured at 2 different time points yesterday to pdf using Octave script and then converted to png.  Images from 2017-12-30 should have transferred now, and QC might be best using the original .bin images.  See below taken at 10:15 and 14:15 (png conversions). 

[image]

[image]

NewcombMaria, 2018-01-02:
Thanks @pless for input during the Tuesday project teleconference.  Much appreciated.  To my understanding of the discussion, these example images are a bit overexposed, however due to the use of a single setting that covers full sun and cloud cover, it's probably a good exposure setting for now.  Future goals: 

- implement a system for different scripts/camera settings for days with different sun/could conditions

- establish a systematic, repeatable method to evaluate settings and exposure decisions:  metrics??  targets deployed in-field??

NewcombMaria, 2018-01-04:
More example converted images from 3 January, mostly cloudy with intermittent sun:
At 12:30
[image]

At 13:30 which must have been cloud-cover: 
[image]

smarshall-bmr, 2018-01-18:
I've been checking some images as well. I'm certainly tempted to back the exposure off now because this calibration was actually set up when the field had 100% cover with sorghum and I was looking to get good imaging down into the canopy. I'm not sure if people processing images would prefer using a single "consensus" exposure time like we are now or if it's better to change the exposure in the early season to avoid overexposing like we are on occasion now.

dlebauer, 2018-01-18:
@pless and @ZongyangLi ^^

NewcombMaria, 2018-04-05:
Later in the season, on a day with passing clouds (15-March), images captured about 20 minutes apart (sun-exposure then cloud-cover).  15-March was also a day with wind. 
[image]

[image]

Paheding, 2018-04-11:
@NewcombMaria  Here is the quality score (QS) for a sample from March 15:
[image]

**QS = 126.13**

Although there are many shadow regions, the image is pretty clear and thus the quality score is within a reasonable range.