terraref/reference-data #216: Revisit data publication requirements

craig-willis, 2018-01-10:
Related to https://github.com/terraref/reference-data/issues/106, meet to discuss publication requirements.

Completion criteria:
* [ ] Finalize requirements

dlebauer, 2018-01-10:
@craig-willis should we meet with someone from the library who can advise us? [Here are notes](https://docs.google.com/document/d/1UtbhumdL9VPGaoVlkos_s2WkUvosZTBwJr2ea1UMCuk/edit) from a previous meeting with Sarah Williams and Heidi Imker about this topic.

craig-willis, 2018-01-19:
Notes from earlier discussions

The original effort focused on Dryad with tentative agreement from the curator to support the remote dataset. However, after further review, we had two major concerns: CC0 license requirement for the data and the need to have a publication in process to submit the dataset.  PANGAE seemed to offer an alternative, but it again isn't clear whether they will support remote data.

I've contacted PANGAE (https://www.pangaea.de/) to see whether they accept remote datasets. PANGAE is hosted in Germany, if that matters.

dlebauer, 2018-01-21:
I don't think it matters that Pangaea is in Germany any more than Nature is in UK. But Dryad's motivation for using cc0 is compelling and I'll run it by the team.

craig-willis, 2018-01-23:
Response from PANGAEA:

>  Das scheint hier ja garnicht die Frage zu sein. Es geht ausschließlich um remote. Wir hatten ähnliche Diskussionen schon mit UFZ und anderen. Ich habe damit keine Probleme, solange wir eine halbwegs brauchbare Einschätzung haben, dass die Daten tatsächlich durch Dritte  nachnutzbar sind und die Einrichtung vertrauenswürdig ist, d.h. die Daten dort auch persistent liegen (offenbar haben sie in dieser Hinsicht ja ein Eigeninteresse).

and in Google-translated English:
> That does not seem to be the question here. It's all about remote. We had similar discussions already with UFZ and others. I have no problems with that, as long as we have a reasonable estimate that the data is actually reusable by third parties and the device is trustworthy, i. E. The data there are also persistent (apparently they have a self-interest in this regard).

dlebauer, 2018-01-23:
Here is Dryad's rationale for using CC0, from http://datadryad.org/pages/faq#info-cc0 see also https://blog.datadryad.org/2011/10/05/why-does-dryad-use-cc0/

All data submitted to Dryad is released to the public domain under Creative Commons Zero (CC0), which reduces legal and technical impediments to the reuse of data by waiving copyright and related rights to the extent permitted by law. In most cases, CC0 does not actually affect the legal status of your data, since facts in and of themselves are not eligible for copyright in most countries (e.g. see [this commentary from Bitlaw](http://www.bitlaw.com/copyright/unprotected.html#ideas) regarding U.S. copyright law).

CC0 does not exempt those who reuse the data from following community norms for scholarly communication, in particular from citation of the original data authors. On the contrary, by removing unenforceable legal barriers, CC0 facilitates the discovery, reuse, and citation of that data. Any publication that makes substantive reuse of the data is expected to cite both the data package and the original publication from which it was derived.

"Community norms can be a much more effective way of encouraging positive behaviour, such as citation, than applying licenses. A well functioning community supports its members in their application of norms, whereas licenses can only be enforced through court action and thus invite people to ignore them when they are confident that this is unlikely." ([Panton Principles FAQ](http://pantonprinciples.org/faq/#q10-what-are-community-norms-and-why-are-they-important))

"…when you federate a query from 50,000 databases (not now, perhaps, but definitely within the 70-year duration of copyright!) will you be liable to a lawsuit if you don’t formally attribute all 50,000 owners?" ([Science Commons Database Protocol FAQ](http://sciencecommons.org/resources/faq/database-protocol/))

Similarly, we do not support the use of licenses that (questionably) assert copyright and add well-meaning, but potentially problematic, restrictions on reuse, such as “non-commercial”, “no derivative” and "share-alike" conditions.

"... given the potential for significantly negative unintended consequences of using copyright, the size of the public domain, and the power of norms inside science, we believe that copyright licenses and contractual restrictions are simply the wrong tool [for data], even if those licenses and contracts are used with the best of intentions." ([Science Commons Database Protocol FAQ](http://sciencecommons.org/resources/faq/database-protocol/))

Dryad’s use of CC0 to make the terms of reuse explicit has some important advantages:

    Interoperability: Since CC0 is both human and machine-readable, other people and indexing services will automatically be able to determine the terms of use.
    Universality: CC0 is a single mechanism that is both global and universal, covering all data and all countries. It is also widely recognized.
    Simplicity: There is no need for humans to make, or respond to, individual data requests, and no need for click-through agreements. This allows more scientists to spend their time doing science.

It is important to note that if you have data that, due to pre-existing agreements, cannot be released under the terms of CC0, please do not submit that data to Dryad. Journals that require data archiving as a condition of publication can make exceptions for such special cases.

craig-willis, 2018-01-23:
Apparently above response from PANGAEA was not meant to be public.  They are discussing supporting this type of dataset internally and will get back to us. We're not the first request, but they don't have any active examples.

dlebauer, 2018-01-23:
I created a repository with a draft manuscript https://github.com/terraref/terraref_data_paper

craig-willis, 2018-01-24:
Closing this as we re-start work on https://github.com/terraref/reference-data/issues/106

closed 2018-01-24 by craig-willis

craig-willis, 2018-01-25:
A final update from PANGAEA:

> Dear Craig, we discussed your request internally. Sorrily, we decided that we cannot publish metadata only. Even if you supply a compilation of content we do not have control on the integrity of the data. Checking the availability of remote files (which we do regularly) is not sufficient. In any case, we do not see a problem to get a DOI e.g. via CrossRef - not accepted by Nature?  Sorry for not having a better answer Michael