terraref/reference-data #217: Check season 2 management dates

dlebauer, 2018-01-18:
From: https://github.com/terraref/traitvis-webapp/pull/15
Any managements records that are not showing up are failing to pass two filters. The first is that the management_id must show up under the managements_treatments table of betydb, where the treatments are taken from the treatment_id attribute of the trait records for that experiment. When I looked into Season 2, I found that the following management ids pass this first filter:

6000000014 6000000001 6000000004 6000000005 6000000006 6000000007 6000000008 6000000009 6000000010 6000000011 6000000012 6000000013 6000000015 6000000016
1471281597 1488843492 1488843492 1488843492 1488843492 1488843492 1488843492 1488843492 1488843492 1488843492 1488843492 1488843492 1488843492 1488843492
6000000002 6000000003

However, none of these managements records pass the second filter, which checks that the 'date' attribute of each management lies within the start and end date of the experiment. That is why no managements records show up for MAC Season 2.

kimberlyh66, 2019-01-04:
I think no managements records are showing up for MAC Season 2 since '%Season 2%' site names are associated with MAC Season 1 experiments. Also, some management ids (1471281597, 1488843492, 6000000002, 6000000003) from above are not found in the managements_treatments table.

kimberlyh66, 2019-01-09:
@dlebauer here are the update statements needed to fix issues occurring with season 2 management records:
```sql
update traits
set treatment_id = 6000000018
where site_id in (select id from sites where sitename like '%Season 2%');
```

```sql
update experiments
set start_date = '2016-07-18'
where name = 'MAC Season 2: Stay-green RILs F10';
```

dlebauer, 2019-01-09:
Thanks - I've run these statements against the terraref instance of the database.

closed 2019-02-19 by dlebauer