terraref/reference-data #218: Mechanism for identifying data errors

dlebauer, 2018-01-18:
For data with known errors:

- add file named "ERROR" to file directories
- can be used for Summer 2017 FLIR as well as VNIR 
- need script like for datasets in date range add ERROR file w/ some content: describe issue; point to github issue(s) if appropriate

dlebauer, 2018-01-18:
Also some dates (Nov 10-12 2016) w/ 0B SWIR datasets: https://github.com/terraref/reference-data/issues/214#issuecomment-358727088

NewcombMaria, 2018-02-02:
Scanner 3D laser data from the overnight scan 2018-01-29/2018-01-30 returned null data files (completely black) when the laser sensor was not enabled in the PLC.  The file directories can be flagged as "ERROR" from 2018-01-29 19:13 to 2018-01-30 04:13.

dlebauer, 2020-02-17:
For FLIR surface_temperature

```sql
begin;
update traits set checked = -1, 
   notes = 'broken FLIR camera https://github.com/terraref/reference-data/issues/190'
	 where variable_id = (select id from variables where name = 'surface_temperature') and
date between '2017-03-15' and '2017-09-30';
commit;
```