terraref/reference-data #219: Query sensor data file path from using terrautils

Paheding, 2018-01-18:
_PLEASE USE THIS TEMPLATE when submitting a new issue_

 
### Completion Criteria

max-zilla, 2018-01-19:
@jterstriep in the error above we just don't have a stream RGB GeoTIFFs Datasets for the 'MAC Field Scanner Season 1 Field Plot 515' plot, but we aren't catching that.

it works for valid plots in later seasons:
```
>>> from terrautils.products import get_file_listing
>>> l = get_file_listing(None, "http://terraref.ncsa.illinois.edu/clowder/", "Pb3AUSqnUw", "RGB GeoTIFFs Datasets", "MAC Field Scanner Season 4 Range 26 Column 6")
>>> len(l)
5
>>> l[0]
{u'contentType': u'image/tiff', u'filepath': u'/home/clowder/sites/ua-mac/Level_1/rgb_geotiff/2017-07-03/2017-07-03__11-48-50-648/rgb_geotiff_L1_ua-mac_2017-07-03__11-48-50-648_right.tif', u'filename': u'rgb_geotiff_L1_ua-mac_2017-07-03__11-48-50-648_right.tif', u'date-created': u'Wed Nov 01 09:52:32 CDT 2017', u'id': u'59f9dfb04f0cfb5041110b98', u'size': u'24470750'}
```

Paheding, 2018-01-19:
From [https://github.com/terraref/tutorials/blob/master/sensors/06-list-datasets-by-plot.md](url)
It looks like the currently available sensors are:

IR Surface Temperature
Thermal IR GeoTIFFs Datasets
flirIrCamera Datasets
(EL) sensor_weather_station
Irrigation Observations
Canopy Cover
Energy Farm Observations SE
(EL) sensor_par
scanner3DTop Datasets
Weather Observations
Energy Farm Observations NE
RGB GeoTIFFs Datasets
(EL) sensor_co2
stereoTop Datasets
Energy Farm Observations CEN

From the error we got I assume the dataset associated with sitename is limited for these available sensor, or maybe there is other reasons? 
Meanwhile, if  VNIR and SWIR sensor datasets could be included in this list would be very useful.

max-zilla, 2018-02-01:
We need to add an explicit message to indicate if files are missing (e.g. some Season 1 RGB GeoTIFFs) so it's clear what's an error and what is unavailable.

either @jterstriep or I should implement in terrautils.

max-zilla, 2018-02-20:
@Paheding pull request is here, if you are satisfied with it we can merge:
https://github.com/terraref/terrautils/pull/44

```
>>> from terrautils.products import get_file_listing

>>> l = get_file_listing(None, "http://terraref.ncsa.illinois.edu/clowder/", "KEY", "RGB GeoTIFFs Datasets", "'MAC Field Scanner Season 1 Field Plot 515'")

>>> l
[]
```

Paheding, 2018-02-20:
@max-zilla This is great. Thanks. We can merge.

closed 2018-02-21 by max-zilla