terraref/reference-data #220: SWIR metadata and focus setting

NewcombMaria, 2018-01-19:
Currently the SWIR metadata doesn't have an obvious option for describing manual changes to the focus setting.  The json metadata files have the following descriptors related to focus: 

optics focal length [mm]:  "25" 
optics focus setting: "2m"
optics focus apperture:  "2.0"   (metadata spelling) 

Earlier this season @smarshall-bmr manually adjusted the focus on the SWIR camera itself (a physical adjustment with no obvious markings to note the new focus point on the camera).  We'd like to be able to document this change.  The change was made before the hyperspectral scan on 2017-12-06.  

2017-12-04 (old focus, clearly out of focus in the images we can view directly from the cache server)
2017-12-06 (new and improved focus) 

@dlebauer suggested we open discussion to determine optimal metadata fields and then have a single metadata cleaning function.

smarshall-bmr, 2018-01-19:
The refocus was part of the preseason cleaning and qc. Maybe we should include a "last service date" metadata field?

dlebauer, 2018-01-24:
That is a good idea to add a last service date to the fixed metadata - could you add the information to the SWIR fixed metadata and create a pull request? https://github.com/terraref/sensor-metadata/edit/master/sensors/SWIR/sensor_fixed_metadata.json 

As long as you add and do not edit fields, the downstream pipeline shouldn't be affected.

dlebauer, 2018-01-24:
@smarshall-bmr do you know when the camera went out of focus?