terraref/reference-data #223: Data publication

craig-willis, 2018-01-24:
This epic covers the data and paper publication.

max-zilla, 2019-02-28:
Last comment from @dlebauer : 
I think that the next step after finishing the current reprocessing is to generate the data package for publication https://github.com/terraref/reference-data/issues/106. Then it will be possible to begin writing up the data paper. During writing / before publication it will be necessary to update field and computational protocols.

max-zilla, 2019-08-01:
@robkooper recommends look at using ORE specification for our JSON catalog:
http://www.openarchives.org/ore/0.9/jsonld

...as well as down the road implementing "holey bagit" approach to allow downloading of TERRA as a zip file with URL files instead of real bytes but otherwise complete.

craig-willis, 2019-08-01:
This is an interesting suggestion, as we've gone with something similar for publishing in Whole Tale. We adopted the [Research Object Bundle](https://researchobject.github.io/specifications/bundle/) with [BagIt serialization](https://github.com/ResearchObject/bagit-ro) and are using the [BDBag](https://github.com/fair-research/bdbag) utilities, since it should support fetching via Globus.  RO-Bundle is based in part on ORE although it seems to be moving toward the new [RO-Crate](https://researchobject.github.io/ro-crate/) format (these things are forever moving).