terraref/reference-data #225: operational checklists, stereo maintenance logs, etc

dlebauer, 2018-01-25:
Need to consolidate spreadsheets, forms for gantry-side logging 

* field book
* daily operational check list
* sensor maintenance

For each of these, define what information these need to contain (can be drafted as spreadsheet or form) .

Completion Criteria

* [ ] define requirements
* [ ] draft forms or spreadsheets
* [ ] discuss & revise
* [ ] implement
* [ ] (optional ... import historical data)

NewcombMaria, 2018-01-26:
I can work on this with Stuart.  

Requirements:  
- The daily operational checklist is genuinely a 'list to be checked' daily to determine if components of the gantry and the camera box are functioning or showing signs of wear.  Purpose: early detection and documentation of issues that need attention.  

- Camera/sensor maintenance log:  issues that are found when completing the operational checklist can be entered into a maintenance log, along with preventative maintenance and calibration etc.

- A while back (>1 yr) we used to complete the Field-Scanner Operations Log for scans and issues that occurred during scans (weather, motion faults, etc)
 (https://docs.google.com/forms/d/e/1FAIpQLSeq9E83vJFCORaGj6TCrq9QUYQLBbAByKCdc_B9Dea8WXyGGg/viewform?c=0&w=1).  Is there still a need for a similar log?  Or a log of new scripts that are developed? 

- Field book:  is a log of hand-measurements and field activities including uav and tractor scans.

NewcombMaria, 2018-02-26:
@max-zilla and @dlebauer I'm been manually updating 2 spreadsheet logs since Feb-13 to cover daily operational checks and a log of issues/problems by day.  Please review when you get a chance and let me know if these formats will be useful on your end.  

The daily checklist responses are categorical:  'checked' (implies operational status); 'not checked' (no info for that day); 'last checked ----' (reports the most recent day when the status was checked); or 'see log' (when there is a problem or issue that needs description it's written up in the checklist log. 

Field scanner operational daily checklist: 
https://docs.google.com/spreadsheets/d/1Ac3o3Psbu8RthmikX_zm6oEnjXj9hit6XGYR2S6fQxI/edit#gid=1899754381 
Field scanner operational log (updated daily): 
https://docs.google.com/spreadsheets/d/1FGRyCixMDddw7xhJdjbmAcawwJra2UcbLtpM7A16oAQ/edit#gid=1899754381

NewcombMaria, 2018-05-05:
@dlebauer and @craig-willis, following up on 3-May discussion I've created a Scan log and also started a log of Field activities in a new document and new format. 

Scan log:  https://docs.google.com/spreadsheets/d/1dZYE6UbF9rmYuUNLkpUNJzXkc7kHPAZhoBxQtGuCUXc/edit#gid=1793149806

Field activities log:  
https://docs.google.com/spreadsheets/d/1y5HqfuTm9OUNdST8UH2vcAiRjMgz2sWt1VoAqVfIe6I/edit#gid=425706872 

Especially for the Scan log, let me know if more or different info would be useful.  The 'good data sets' in the last column are from the automated emails and having the info in one spreadsheet will help me compare from day to day to see if any images are dropping from the scans.

dlebauer, 2018-05-06:
Thanks - it looks great, I've made a few edits to use a more standardized date-time format

YYYY-MM-DD HH:MM

closed 2018-05-10 by dlebauer