terraref/reference-data #226: SWIR and VNIR metadata errors need back-correction for X axis position

NewcombMaria, 2018-01-30:
Yesterday 29-Jan Stuart, John and I spent time examining hyperspectral data and importantly also examining the associated metadata.  We identified consistent metadata errors in the X axis position and also the timestamp for the hyperspectral data files.  It appears the X position is recorded at the previous short-axis scan (Y-axis), before the camera box actually moves to the actual X-axis position where it starts the data line-scanning swath.  

Since both hyperspectral cameras are running in un-triggered mode, there is a single X axis location and a single timestamp for the entire short-axis (Y axis) pass across the field.  Scripting the hyperspectral scan is complicated when running in the un-triggered mode.  It turns out that the script was programmed in a way that the hyperspectral measurements actually begin at the location of the end of the previous short-axis pass, with the associated metadata file capturing the timestamp and the X axis position also at the end of the previous pass, not at the actual position/time of the data collection pass.

We spot-checked a couple 2017 data and metadata files that are still on the cache server and from our first look it's likely that all of the summer 2017 SWIR data might need metadata X position correction. (There were 3 scripts during 2017:  Day1 and Day2 scripts to cover alternating ranges of the field to be able to cover all ranges over 2 days, and also a "diurnal 5-range scan" during the August photosynthesis field campaign.)  

Defining the scans and dates that will need back-correction is a work in process.  We will complete more spot checks and update the issue as more details are know.

dlebauer, 2018-01-31:
@NewcombMaria a few questions:

* in the issue above there is no mention of the VNIR, although it is mentioned in the issue title. Does this apply to both?
* what is the process that you are using to identify if there is an issue? Is there a particular metadata field or other information that we can search for?

NewcombMaria, 2018-01-31:
@dlebauer we are still in process of doing spot checks to determine which scans have X position errors (and associated timestamp errors) and whether it's always both SWIR and VNIR or maybe only SWIR for some scans.  Also Stuart will look at the script code for S4 scans to see when/where measurements were programmed to start.  This is what is know so far: 

- Season 5 (current wheat experiment):  the scan that covers 3 ranges it definitely applies to both VNIR and SWIR. 

- VNIR data has been removed from the local cache server for dates before October 2017, so I'm not going to be able to spot check VNIR from S4 sorghum 2017 from here.  SWIR data goes back to June 2017 on our local cache so I can look at S4 SWIR files.  Stuart might be able to get a good indication from the script code, but it would still be good to spotcheck the metadata itself.  

- We don't have a simple process for identifying the problem data-sets worked out yet.  At first we thought we could compare the StereoTop metadata with the swir/vnir metadata, but since the timestamp is off whenever the x position is off it gets complicated.  When we have a chance to look carefully at the swir 2017 S4 files and also the script code we will be able to come up with a repeatable process for identifying the problem.

NewcombMaria, 2018-02-11:
@craig-willis I've figured out a method worth testing to determine in high-throughput if the SWIR and VNIR x-axis positions are correct or not, with help on your end.  If this method seems reasonable, it could test it on hyperspectral datasets from this season (S5), starting end of November through current date.  It involves taking select information from the hyperpectral metadata json files, hyperspectral frameIndex.txt files and StereoTop metadata json files for the same scan names on the same day. 

Each SWIR and VNIR data folder includes a frameIndex.txt file, with timestamps recorded across the whole data measurement.  Would it be possible to write a script or batch process to do the following for each hyperspectral data folder (for now starting with Season 5 data)?: 

1. From the SWIR or VNIR frameIndex.txt file extract the mid-point or median time, or any time near the middle or just past the middle of the measurement

2. From the associated SWIR or VNIR json metadata file extract the x-axis position

3. Search the StereoTop timestamps in the json metadata files for the time closest to the frameIndex midpoint time, and extract the x-axis position from the StereoTop metadata file.  This x-axis position associated with the StereoTop measurement time is accurate

4. Compare the hyperspectral x-axis position recorded in the metadata file to the correct x-axis position found by looking up Stereo image taken at the mid-point time of the hyperspectral measurement.  If the x-positions match, no revisions are required.  If the x-axis positions differ, the correct one is the StereoTop metadata x-axis position. 

Does this make sense?  Is it practical?  Let me know if you have questions.

jdemieville-ua, 2018-07-25:
@NewcombMaria  and I have talked through a method of figuring out the correct positions. Included is a rough diagram.

[image]

1. Add the start time of each hyperspectral measurement to a matrix of length n.

**t_hyperspectral = [t_0, t_1, t_2, …]**

2. Compute a matrix of time differences with length n-1.

**t_hyperspectral_differences = [t_1 – t_0, t_2 – t_1, …]**

3. Divide each element of the differences matrix by 2, and then add it to the corresponding start time. This should give a time corresponding to near the center of each pass. It generally won’t be the exact center, but it will be close enough to be away from the field edges.

**t_hyperspectral_midpoints = [t_0 + ((t_1 – t_0) / 2), t_1 + ((t_2 – t_1) / 2), …]**

4. Develop a multidimensional reference matrix composed of x-positions from the StereoRGB and corresponding timestamps for those RGB measurements. 

**reference =** 

**x_1 | t_11, t_12, t_13, t_14, …, t_1n**
**x_2 | t_21, t_22, t_23, t_24, …, t_2n**
**x_3 | t_31, t_32, t_33, t_34, …, t_3n**
**… | …**
**x_m | t_m1, t_m2, t_m3, t_m4, …, t_mn**

5. Compare each hyperspectral midpoint time to the reference matrix. If the hyperspectral midpoint time falls within a range encompassed by an x-value, then assign that x-value for the hyperspectral. It might be easiest to convert each time to a simple format for mathematical operation, such as Unix time. 

**if (t_hyperspectral_midpoints[0] >= t_11) AND (t_hyperspectral_midpoints[0] <= t_11)
	then x_hyperspectral[0] = x_1
else 
	repeat for next set of boundary conditions**

This logic could be ran against all timestamps recorded for the Stereo RGB to assign appropriate x-positions to the hyperspectral data. As the final pass would not have any subsequent scan start, it would be necessary to explicitly state that its position matches that of the final RGB measurement.
My thoughts are that this would allow us to avoid:
•	Looking at non-metadata files,
•	Looking at pre-correction positions of hyperspectral,
•	Trying to find the exact matches for timestamps.

dlebauer, 2018-07-25:
A few thoughts - 

* First, it would be useful to look at the metadata inside the files (as opposed to the json files).
* It would also be useful to overlay the hyperspectral and RGB images to see if the interpretation is correct.
* the hyperspectral data have four dimensions: x, y, t, and wavelength. The x and t dimensions are the same length - each ~1mm scan line along the x axis has a distinct time point (and min (t) would represent the start time).
* The time dimension is stored in units of ‘days since 1970-01-01’ (like Unix time but with days instead of seconds)

Would it be sufficient to find a time point in the hyperspectral data that matches the time an RGB image was taken, then use the position of the sensor box at the time the image was taken to correct the hyperspectral data? Is the offset static, or did it change over time?

This is a lot to digest - so I apologize if I have missed something, but it would be useful to have an example with actual data that we can work through, generate automated tests for the types of errors that are present, and then a script to correct the dimensions; the function ncap2 in the nco library makes it quick and easy to transform dimensions (something like `ncap2 -d x=x+5,y=y-2 output.nc`, but see [ncap2 documentation](
http://nco.sourceforge.net/nco.html#Syntax-of-ncap2-statements).

NewcombMaria, 2018-09-07:
@dlebauer are there any updates?  Are x-position data being corrected during re-processing, or are there any flags for data users that the x-position for VNIR/SWIR may be incorrect?  There's a good chance that any/all VNIR/SWIR data from summer 2017 can be used as an example, but I'm not certain of script changes during season4.