terraref/reference-data #228: Obsolete PlantCV extractors directories to delete

dlebauer, 2018-02-06:
@nfahlgren please confirm that the following should be deleted:

Under /sites/danforth:

* sites/danforth/derived_data
* raw_data/
   * helloMax
   * jdtest
    * test_snapshot_lt1
    * snapshot295351 (?)
* Level_1/plantcv/ <- not clear what level of hierarchy these should be in
  * lots of two-letter directories (de, df, e0, e1, e2, .... ???) that contain 4-5 snapshots each
  * PlantCV Extractor Demo
  * test
  * ddpscIndoorSuite - snapshot* <- lots of these ... should these be nested under sorghum_pilot_dataset?
  * TM015_5_051616 <- looks ok but correct level?
  * TM015_5_052716<- looks ok but correct level?

nfahlgren, 2018-02-06:
@dlebauer I think everything you list can be deleted. I'm not sure what process created the two-letter directories but the ones I looked at were empty. The ddpscIndoorSuite data can be deleted, I don't think we need the pilot data anymore.

The TM015_5_051616 and TM015_5_052716 directories contain the outputs from Clowder for PlantCV run on each input dataset. We have the raw data organized into the same folders so I assume that level of hierarchy is okay?

dlebauer, 2018-02-06:
If we delete all of the ddpscIndoorSuite data and the two-letter directories then the hierarchy makes sense. The exception is that `raw_data/` outputs go to `Level_1/plantcv`, but this isn't important if it makes sense to you. 

To confirm:

```sh
cd terraref/sites/danforth/
rm -rf derived_data
cd raw_data/
rm -rf helloMax jdtest test_snapshot_lt1 snapshot295351 

cd ../Level_1/plantcv/ 
rm -rf "PlantCV Extractor Demo", test 
rm -rf ddpscIndoorSuite - snapshot* 
rm -rf [abcdef]*
```

keep:

```
raw_data/sorghum_pilot_dataset
raw_data/TM015_5_051616 
raw_data/TM015_5_052716
Level_1/plantcv/sorghum_pilot_dataset
Level_1/plantcv/TM015_5_051616 
Level_1/plantcv/TM015_5_052716
```

dlebauer, 2018-02-06:
@nfahlgren does the above look okay? If so, I can ask @jdmaloney to delete

nfahlgren, 2018-02-06:
@dlebauer yeah, the plantcv directory is probably a bit unnecessary, but if we ever processed the Bellwether data with a second method then it would make sense. So I guess it's future proof :)

The above list of deletes and keeps looks correct to me.

dlebauer, 2018-02-07:
@jdmaloney can you delete these?

max-zilla, 2018-02-08:
These were deleted: https://github.com/terraref/reference-data/issues/228

dlebauer, 2018-03-06:
still need to `rm -rf terraref/sites/danforth/derived_data`

dlebauer, 2018-03-06:
looks like a bunch of other folders still exist (at least looking on Globus)

[image]