terraref/reference-data #231: Read VNIR raw data

ebabaeian, 2018-03-01:
Hi ,
I have a problem with extracting VNIR data. When I download a VNIR raw file with ~ 28GB size and extract the compressed (tar) image I get a smaller size as ~8GB. Also this 8GB image has jpg format and when I open it in a software like ArcGIS I see a stretched image. Any thoughts for this? Thanks very much.
Best Regards
Ebrahim

 

_PLEASE USE THIS TEMPLATE when submitting a new issue_

### Description (REQUIRED)

 
### Completion Criteria

dlebauer, 2018-03-01:
Could you point to a specific file? Does this happen on multiple dates or just one? Could you post (copy-paste) a screen shot?

NewcombMaria, 2018-03-01:
@ebabaeian and @dlebauer the date of the VNIR file is important.  VNIR data from 2018-11-28 until about 2018-02-5 are good quality (as far as we know).  After about 2018-02-05 the VNIR camera started having problems and the data cubes are poor quality.  Headwall has been contacted and we will probably need to ship the camera off for repairs. Ebrahim, please let David know which date you are looking at when you post a screen shot.

Paheding, 2018-03-01:
@ebabaeian @NewcombMaria For measure the image quality, I just uploaded an extractor yesterday that is designed to evaluate the quality of images with different spectral bands such as VNIR, SWIR, etc. The extractor will output a warning if the quality is below a threshold. The related Github issue: https://github.com/terraref/computing-pipeline/issues/413
Currently, we are trying to configure this extractor and get it running in the pipeline. Once it is fully functioned, it should be helpful to tracking the image quality and explore the quality issues.

ebabaeian, 2018-03-02:
Hi @dlebauer @NewcombMaria 
The image date that I downloaded was Feb 1, 2018. The date that we have SWIR data throughout the field. I only downloaded 3 VNIR swath associated with the location of the TDR sensors for soil moisture. All these three VNIR swaths are stretchered. The file format is "jpg' but with ~8GB size which is attached. 

[image]

Thanks
Ebrahim

dlebauer, 2020-06-05:
closing - hyperspectral pipeline is being refactored.

closed 2020-06-05 by dlebauer