terraref/reference-data #236: bety not dumping

dlebauer, 2018-03-08:
I was trying to debug an issue with the bety.terraref server that is used as a clone of bety6 for the traitvis webapp and for workbench users to access a read-only copy of the trait data.

The database wasn't updating (the most recent 'updated_at' was Nov 2017) so I tried to re-populate the bety.terraref database manually. This didn't help. So I went into bety6 and checked the dump in /var/www/html/bety/dump/bety6/bety.tar.gz. This (and ../bety0/bety.tar.gz) haven't been updated since Dec 8.

## Completion Criteria

* [ ] debug error in dump script
* [ ] write test to send alert if the dump script fails or otherwise doesn't run daily

robkooper, 2018-03-12:
Updated dump script, was using old argument that is no longer supported and would ignore rest of arguments. Now dumping correctly again.

Changed so that cronjob will send email to me when it fails, however this would not catch the errors above since it was silently ignoring command line arguments.

robkooper, 2018-03-12:
Monitoring could be fixed by using the [PEcAn monitoring page](http://pecan-dev.ncsa.illinois.edu/pecan/status.html).

closed 2018-04-25 by dlebauer