terraref/reference-data #239: Daily scan of standard calibration objects

dlebauer, 2018-04-05:
Our goal is to provide a daily sanity check and reference for all of the sensors.

Starting in the 2018 Sorghum season we will reserve one plot for calibration targets. This can be in the corner of the field and scanned prior to and optionally after each run.

* Choose one plot for the calibration objects
* place calibration targets in the plot each day and scan
    * these should be 2m below the scanner box
* Define the boundaries of the calibration panel plot and add to BETYdb 

Calibration targets to deploy:
* [SphereOptics Diffuse Reflectance Standards](https://terraref.ncsa.illinois.edu/clowder/datasets/5873a9434f0cad7d8131b29a?space=594bda394f0cf8bbab271dfb)
* [Aluminum Checkerboard]()
* [LabSphere Spectralon Color Standards](https://terraref.ncsa.illinois.edu/clowder/datasets/5873a8184f0cad7d8131a4dd)
* Temperature target? 

### Completion Criteria

* [ ] Develop Protocol
* [ ] Implement
* [ ] Document (in documentation or protocols.io)

NewcombMaria, 2018-04-12:
I've started a document in discussion with @JeffWhiteAZ that outlines specifics to be figured out: 
 dimensions of the platform required to deploy the targets, and selection of targets, etc.  The document is available for review: 
 https://docs.google.com/document/d/1NoeTRJZ-bxmeJUReBE_5Anmz8Rey6UeU6IrOV9aTAto/edit

NewcombMaria, 2018-04-29:
We have a set of targets assembled focusing on the stereo-RGB for now until the SWIR, VNIR, and FLIR are back from repairs.  Comments and input on the targets are welcome.  They fit on a 40cm by 80 cm platform. 
The plan is to deploy the targets on the west side of Range 52 during the day scans.  The image includes a couple small potted plants to help test exposure.  
[image]

NewcombMaria, 2018-05-02:
Here's a PDF conversion of the image capturing today's scan of the targets, after slight adjustments to camera settings.  Coordinates X = 202.3, Y = 20.5.  A few seedlings show up in the image.  

Comments/input on settings called from the script (exposure, gain, focus-by-height adjustment) are most welcome.  Aperture is a manual setting.  Current setting "optics aperture setting (both)" = 6.7. 
[image]

dlebauer, 2018-05-03:
@Paheding and @pless  what to add to evaluate focus? ... something like 

[image]

As of May 1, the targets are the plot named "MAC Field Scanner Season 5 Range 52 Column 2" (prior to this, they were in Column 3).

Paheding, 2018-05-03:
@dlebauer I can run evaluation in terms of focus, and comparison results over series times.

dlebauer, 2020-02-13:
@jdemieville-ua have you implemented this? Is a protocol documented somewhere?