terraref/reference-data #240: Update 'RAW Data File Count/File Size' report to include MAC weather station

NewcombMaria, 2018-04-18:
For first-level data-collection quality checks it would be helpful to include data files from the MAC weather station (the one on the tripod in the corner of the field) to the automated 'RAW Data File Count/File Size' report.  

Recently we learned that there has been a problem with intermittent weather station logging or data transfer that extends a long time back (periodic data gaps).  And from 2018-02-20 the data logging seems to have stopped.  I'll catch this type of problem sooner if it's included in the File Count/File Size update.  John H. thinks he has fixed the problem and we'll monitor daily for now.  

An example screenshot of the report from 2018-04-04.  The 'stereoEast' and 'stereoWest' lines could be removed.  These instruments have been premanently removed from the gantry side-arms. 

[image]

dlebauer, 2018-05-02:
It would also be nice if the irrigation, weather, and environment logger showed up on the terra-logging interface http://terra-logging.ncsa.illinois.edu:3000/d/000000002/sensor-file-counts-and-size?from=now-6M&to=now&orgId=1

jdmaloney, 2018-05-04:
This is getting knocked out today.  I have stereoEast and stereoWest cleaned up already, will be getting weather added in and the changes made on the terra-logging site.

NewcombMaria, 2018-05-04:
Excellent.  Thanks.  For the MAC weather station data (corner of field) there is a pattern of gaps in the cache data folders that I don't understand.  From what I can tell, data are being continuously collected, but in the cache folders there are days with only a subset of data and then a day with no data.  I think something like 20-April and 27-April are empty folders, but most days in between appear fine.

jdmaloney, 2018-05-08:
@NewcombMaria This is in the report now and @dlebauer on the dashboard too.  I re-ran this morning's report (hence the email that just went out) and things look good.  I'm closing this issue.

closed 2018-05-08 by jdmaloney