terraref/reference-data #242: Environment logger Geostreams met data should be 5 min (not 5s) frequencies; netcdf should be available on thredds

dlebauer, 2018-05-17:
Per specification in https://github.com/terraref/computing-pipeline/issues/156#issuecomment-261286229 the geostreams met should have 5 min frequency. Currently the environment logger has 5 s frequency.

The high frequency data can be accessed in the environment logger netcdf files (@robkooper these should be available on the thredds server).

robkooper, 2018-05-18:
What is the path for these files on disk, I can add them to the script to generate the thredds database.

dlebauer, 2020-06-04:
Timestep fixed in https://github.com/terraref/computing-pipeline/issues/156#issuecomment-261286229 e.g. https://terraref.ncsa.illinois.edu/clowder/api/geostreams/datapoints?stream_id=46431&since=2017-01-02&until=2017-01-31

thredds no longer a priority

closed 2020-06-04 by dlebauer