terraref/reference-data #243: First-pass QC for 2017 scan for stomatal regulation campaign

NewcombMaria, 2018-05-22:
@max-zilla following up on the Tue telecon discussion on field campaigns, it would be great to be able to view a full-field overview of the 2017 scan that ran with the box raised high with larger FOV for purpose of being able to cover the field quickly for thermal data.  The name of the script file is 'StomatalRegulation_Sorghum4_ranges0.33m.cs'.  I don't know exactly which dates, but likely dates to check include: 
2017-06-16
2017-06-19
2017-06-21
This will help us with rough QC to determine if this scan could provide valuable data for this current season.

max-zilla, 2018-05-29:
The higher resolutions for 06/21 are still being generated, but the thumbnail is complete and on disk under ua-mac/Level_1/fullfield/2017-06-21 (see stomatalregulation TIF).

[image]

There are gaps between the photos. The higher resolutions might make this more obvious, but if we expected full coverage I will check the metadata to see whether the FOV field we use for georegistration actually reflects the larger value or not.

max-zilla, 2018-05-29:
talked with @craig-willis about this. i was reminded that we store the camera FOV in fixed metadata in clowder as something that was not expected to change (like position in camera box, for example) so changed FOV settings, while they may be present in metadata.json files, are currently fetched from fixed md for processing. we will have to discuss refactoring pipeline to support variable FOV and reprocessing any days where this would've been changed.

craig-willis, 2018-05-31:
@max-zilla Please create follow-up issue to address FOV problem and close this if possible.

max-zilla, 2018-06-13:
Discussion has moved here: https://github.com/terraref/computing-pipeline/issues/456

closed 2018-06-13 by max-zilla