terraref/reference-data #244: Changing RGB exposure mid-scan

max-zilla, 2018-07-02:
This is a topic to discuss the idea of adjusting exposure during a scan.

The exposure of RGB camera has been changed based on sunlight under the box for quite a while, but not mid-scan. @ZongyangLi and @craig-willis and I need to determine what impact this could have on downstream extractors.

The exposure value is stored in variable metadata for RGB datasets already.

max-zilla, 2018-07-02:
@NewcombMaria mentions there are other github issues with Pless et al discussing these exposure values.

max-zilla, 2018-07-02:
Consistent exposure during morning RGB scans with shade under the box, but for FLIR scans (sun settings) exposure is turned down.

There are ~3 different exposure categories currently used. 

on 07/01, ran two with shade setting and two with sun settings. Script name indicates 'shade' or 'sun' historically.

NewcombMaria, 2019-01-09:
@max-zilla and @ZongyangLi as you work with the RGB extractors and StereoTop images, it's a good time to return to this issue of changing the exposure setting during a scan for situations when the cloud and/or lighting conditions change.  Recently @jdemieville-ua and I have been talking about how to select the exposure setting for a scan on a day when it starts out cloudy but transitions to more sun before the end of the scan.  We would get the best image quality results if we can adjust exposure as the light condition changes, but we don't want to complicate or break any downstream processing.