terraref/reference-data #247: FLIR Image Deterioration Mid-Scan

jdemieville-ua, 2018-08-10:
FLIR QC spot-checking shows deterioration mid-scan beginning:
24 July - 12:52:54
25 July - 12:21:32
27 July - 14:52:48
On the morning of the 27th, we tried the camera and imagery appeared functional.

[image]
[image]

We suspect it may have to do with temperature. FLIR specifications state a maximum operating temperature of 50 deg C. On the 24th, 25th, and 27th, camera temperatures were approximately 55, 58, and 55 deg C, respectively. Ambient temperatures had daily highs approaching 47 deg C. The FLIR is one of the devices without any active cooling.

We are seeking a method to check FLIR metadata for the following:
1) Frequency of camera temperatures (lens/shutter/front) exceeding 323 K,
2) Date/Time ranges corresponding to camera temperatures exceeding 323 K, and
3) Ambient air temperatues (either from gantry or field sensors) corresponding to the times of elevated camera temperature. 

We also are curious if the problems we see are manifested in the fullfield products. I don't see any fullfield imagery past July 20th in Globus. 

What would be the most efficient method for us to get this information?

max-zilla, 2018-08-14:
july 23 & 24 fullfield is now available in Globus. other dates in process.

NewcombMaria, 2018-08-21:
We've discussed the issue with Jeff White and Andy French.  Jeff reviewed the consecutive single images from scan on 7/25:  12:21:18 (good image) and 12:21:32 (deteriorated image). 
 
Email from @JeffWhiteAZ: "when converted to Celsius with Andy’s coefficients for 35oC, there is a single pixel at x=343, y=259 with a value of -135.0474oC. In the raw bin, this pixel has a value of 1.  That distorts the temperature scale as seen in 780858…png.  So the question is whether this is a bad pixel on the sensor or something else."

dlebauer, 2020-06-25:
In the end, can we conclude that the 'deteroration' is really due to a single bad pixel?

jdemieville-ua, 2020-06-25:
@dlebauer : If I recall correctly, that was the conclusion. There was discussion of changing the value of a bad pixel by applying something like a neighborhood average filter if a pixel value was outside the realm of reality, but I have no information on whether that was ever implemented. It's such a small percentage of the total image area that it doesn't make sense to throw out the data. This sort of correction should be able to be applied easily by someone familiar with the conversion between the bin files and tiffs. 

The issue was intermittent and appeared to be related to high temperatures. It remains to be seen whether the issue reappears now that the camera is climate-controlled within the rated operating range.