terraref/reference-data #248: Test scans for input on VNIR settings

NewcombMaria, 2018-08-23:
Today @jdemieville-ua and I ran multiple VNIR test scans across the short axis of the field with targets deployed at a range of exposure settings, all at 0.04 m/sec velocity (y axis speed).  The majority of the test runs were in triggered mode, one in untriggered mode.  Exposures ranged from 15 to 35 milliseconds.  Conditions were mostly cloudy during our test scan, but we had to proceed without waiting for a sunny day.  We can repeat the tests again on day without clouds after planting.  Photo copied below with the primary targets labeled.  A couple potted sorghum plants were included.   

@remotesensinglab would you be able to advise us on settings?  
Let me know if you need additional information. 

[image]

Paheding, 2018-08-23:
@NewcombMaria and @jdemieville-ua Thanks for running the test scans. Were that VNIR data stored in Globus? If yes, could you please let me know the folder names? So that we can look into these specific data and find out the proper settings.

jdemieville-ua, 2018-08-23:
@Paheding : It doesn't look like they've made it into Globus yet. I'll keep an eye on it and update you with a list of directories when they're available.

max-zilla, 2018-08-23:
This issue is why they haven't been transferred via Globus yet:
https://github.com/terraref/computing-pipeline/issues/492

We have a ticket open with Globus, the endpoint is reporting an expired certificate even though JD looked on the cache server in AZ and it reports the cert won't expire until 2021. until this is resolved, transfers can't occur.

Paheding, 2018-08-23:
@jdemieville-ua and @max-zilla Thanks for the info. I look forward to seeing the images.

remotesensinglab, 2018-08-28:
All,
Patrick and I evaluated the data. We could run the quality check pipeline code on these as well to see if we can warn the quality issues automatically. But wanted to provide a quick note as we prepare for data collection. The report is available here.
[HSI_evaluation.pdf](https://github.com/terraref/reference-data/files/2327844/HSI_evaluation.pdf)

I think the 20 ms exposure time with 50 frame rate is the best among the settings:
(1) Reflectance at NIR is under or equal to 0.55 which is expected for plants
(2) reflectance for 95% panel is where it is supposed to be

THe remaining issue is there is still distortion on the data. Pixels is not square. It is not a major and can be fixed by changing the frame rate. It may be fixed by either increasing speed a little bit or decreasing the frame rate. 

Vasit

max-zilla, 2018-08-28:
@Paheding here are the datasets we transferred for VNIR from 8/18 and 8/20:

[image]

[image]

...this matches the files and sizes on the gantry side. For the 20th the raw data goes from 30 GB on the first image to 20 MB on the last, but that matches the source.

remotesensinglab, 2018-08-28:
Ok. We will review these ASAP and report back to the team. thanks

rjstrand, 2018-08-28:
With respect to triggered acquisition, we have no real frame rate.  Frames are triggered based on position.  Position is based on the motion of the camera box.  We can change the trigger rate by either changing the velocity of the camera box, or the division rate in the frequency divider on the trigger circuit.

The Headwall frame period setting has no impact on the triggered acquisition.  Nevertheless, it must be set to a value that is larger than the exposure time.

I will verity with Headwall, but I believe that exposure time cannot simply be set to a value of 1/trigger rate.  There has to be some "processing" time between triggers.  This is why we ran the VNIR with with a 35 ms exposure at a 50 ms trigger rate.

The y-axis encoder signal is currently divide down so that the cameras trigger at a rate of approximately 1 scan per 0.95 mm (As I remember, currently dividing primary pulse train by 65).  We can modify this by changing the divisor on the frequency divider.

- Bob

remotesensinglab, 2018-08-28:
Hi Bob,
Do not change anything until we evaluate other datasets. It seems there are more data than we evaluated that have been transferred to the server recently. 

We will look at all systematically and get back to the team next day or two.

Thanks

Vasit

Paheding, 2018-08-30:
Hi all,

We have run the image quality evaluation algorithm on the available VNIR test scans, and conducted some analysis on the spectral profiles of plant and white pannel. All results are attached here ([HSI_evaluation.pdf](https://github.com/terraref/reference-data/files/2337821/HSI_evaluation.pdf)). For convenience, a short summary is as follows:

- **Suggested Setting**: 20 ms exposure time with 50 frame rate in trigger mode. Note that we selected 20 ms among three different values (i.e., 20ms, 25ms, and 35ms ) based on the VNIR data availablity in Globus. See a comparison chart below:

[image]

- The reflectance of plants looks normal (see blow).
- The reflectance for 95% white panel also looks okay (see blow).

[image]

**Remaining issue**:  The geometrical distortion on the pixels, see the shapes of plant and color board in the following image:

[image]

dlebauer, 2018-08-31:
Regarding the geometrical distortion - is this file on globus where I can take a look? If you are referring to the fact that the round objects appear oval, isn't this just a reflection of the fact that the size of the pixel in the left-right direction is determined by the camera and the scan direction (up) is controlled by the speed of the camera? i.e. is this an artifact of trying to fit rectangular information into a square raster image, and if we have correct geospatial reference does this distortion affect the interpretation of the results?

It isn't clear how the plants are distorted, but we have always had waviness that I've assumed is due to wind blowing the plants back and forth as the scan line is pushed forward.

Other teams have mentioned that geometric calibration is a uniquely challenging task, but that it may not directly impact many of the key features we are trying to measure.

What impacts will this distortion have on biological inference?

dlebauer, 2018-09-01:
a few minor details on the soil mask data products ... the way the files are currently generated, the metadata is incomplete (if you compare the output of `ncdump -h file.nc` on a L1 vnir file compared to this, you will see that each variable in the L1 file has a number of attributes including name, long name, units, description, notes, etc.  Also, the original dimension 'x' and 'y' have been replaced with 'x_dim' and 'y_dim' which is confusing, and in the soil mask, 'x', 'y', 'latitude', and 'logitude' are all empty. 

It would be great to have a soil mask data product, but as it stands this would need some work to clean this up. NCO has a lot of utilities that _should_ make it 'easy' to copy x,y,latitude,longitude from the input file to the output file, delete any obsolete variables, and add appropriate metadata (perhaps the python library does as well).

Paheding, 2018-09-03:
@dlebauer The files are located at Globus-- **/ua-mac/raw_data/VNIR/2018-08-22/**. Typically, the distoration can be due to mirror scans, velocity variance, weather conditions, etc. I am not quite sure if the wind blowing occurs at that date that caused the distoration. Based on the experiences from UAV based hypersepecral image collection, the pixel distoration can be corrected by properly adjusting some parameters such as framerate, flight height, etc.

Paheding, 2018-09-03:
@dlebauer The soil mask extractor (https://github.com/terraref/extractors-sunshade) will process a netCDF data file and export a binary mask of
non-vegeation area from hyperspectral imagery, and generates background (typically soil) 
removed hyperspectral imagery with the associated NDVI value as well as soil fraction area. The georeference points are also added to the meta data. These outputs were dicussed and suggested by team. We can definitely add more variables if they are useful.

Paheding, 2018-09-03:
@NewcombMaria We glad to hear the analytic report is helpful. And thanks again for collecting more data.  Especially the data containing seedling emergence and lighting condition would help to robust analysis on the spatial and spectral content. We will provide more analytic results on the coming data and share with the team.

SeanHartling, 2018-09-10:
All,

Just wanted to share some test data we gathered to examine the exposure settings for hyperspectral collection.  The data below was collected from our UAV platform with Headwall Nano VNIR hyperspectral camera.   Despite our best timing attempts, there were intermittent clouds during the collect.  The two left scenes are from the same flight path.  A-1 is the start of the mission, then A-2 is in the same flight line (arrows indicate direction of travel) and B is next flight line in the opposite direction.  You can see the effect of lighting conditions and exposure between the three scenes.  The right two scenes demonstrate the effect of exposure and lighting conditions on reflectance.  All the reflectance values on the reflectance tarp are higher on scene B compared to scene A.  This should be something to consider when dealing with a fixed exposure rate.  We will continue testing to figure out the best approach for this phenomenon.
[image]

Paheding, 2018-09-11:
@dlebauer Althought the correction on the pixel distoration is not current priority, we have quick-checked several VNIR data that were previously captured, It looks like the spatial resolution is better than recent one.  A sample image is shown below:
 
Image tag: 2018-02-01__15-05-34-160
 [image]

However, we need to check the further upcoming data in case of the pixel distortation is due to the wind blowing or any other weather conditions, as you suggested.  Another possible reason would be some limitations of this new VNIR sensor compared to the previous one.

NewcombMaria, 2018-09-25:
Thanks Patrick and Vasit for input on exposure settings.  Now that the sun angle is lower and there is more sunlight, we are wondering if the 20 ms exposure setting is best for bright sunlight, or if 15 ms exposure for VNIR and SWIR may be better when the image is in full sun.  

We collected data on 2018-09-22 to compare the 20 ms and 15 ms exposure settings.  
SWIR 15 ms:  /ua-mac/raw_data/SWIR/2018-09-22/2018-09-22__13-21-35-977/ 
SWIR 20 ms:  /ua-mac/raw_data/SWIR/2018-09-22/2018-09-22__13-34-50-596/ 

VNIR 15 ms:  /ua-mac/raw_data/VNIR/2018-09-22/2018-09-22__13-21-55-992/ 
VNIR 20 ms:  /ua-mac/raw_data/VNIR/2018-09-22/2018-09-22__13-35-10-611/ 

@remotesensinglab would your group be able to evaluate and let us know which exposure is better now that the sun angle allows more sunlight under the camera box?

Paheding, 2018-09-26:
@NewcombMaria Thanks for conducting experiments. We will evaluate those data and provide the feedback.

Paheding, 2018-10-25:
@NewcombMaria  We have conducted some experments. The results suggest that those two settings do not make too much difference. However, 20 ms may be slightly better for both VNIR and SWIR cases. See the graphs below in details. 

[image]

[image]