terraref/reference-data #249: MAC Season 7 field metadata

NewcombMaria, 2018-09-04:
@dlebauer the site/cultivar information for Season7 field metadata is now located in the shared folder: 

https://drive.google.com/drive/folders/1-6WzS881x_sEAD0gleWLxewXJqPDeXPr 

There are separate files for plots (864 in total) and E/W subplots (1728 in total).  Let me know if there are questions.

dlebauer, 2018-11-26:
@NewcombMaria do you have a list of the cultivars for season 7? I see in the spreadsheet (https://docs.google.com/spreadsheets/d/1iqqSAwiB56w_X1OElE4SvV0vTuxnceM23uDhpkeR0_4/edit#gid=144503171) that there is a list of cultivars, but it is not clear if information like 'bloomless' and 'wildtype' should be appended to the cultivar names or if they would be better put in the notes field, or perhaps the notes field. Also, it would be useful to know where these cultivars came from, and any other metadata like lineage or source that we could provide in the BrAPI interface.

dlebauer, 2018-11-26:
@kimberlyh66 for season 7 it looks like the plots need to be divided into 4 'experiments':

- MAC Season 7: RIL
- MAC Season 7: Uniformity
- MAC Season 7: Border
- MAC Season 7: F2

the links between sites and experiments will be found in the experimental design tables above

dlebauer, 2018-12-03:
from @NewcombMaria: 

One important point about the F2 families is that the F2 generation is still segregating so each plant in the plot is genetically distinct.  This is different from lines included in an association panel (BAP or Durum Wheat Panel), and also different from the F10 generation of a RIL population that is advanced to a generation that is no longer segregating. 

The mutant lines are also different in that the mutation is named, for example "leaf firing" is a trait, but also the name of the mutation that was selected. 

David, I'll revise the metadata site-cultivar file in the shared folder to include the following info, copied from the message from Dr Xin (one of the two researchers who provided the seed): 

"We don’t have ID for each mutant or F2 population.

25M2-0345 is the ID for the line in the mutant library or the ID for the M2 plant that was used to generate the pooled M4 seed pools in the mutant library. 25 is the concentration of EMS used (0.25%), M2 is the generation when we started single seed descent. 0345 is the series number. Dwf/erl are the mutant we observed from the line. Usually, each line harbors multiple mutations. If the two mutant phenotypes are observed on a single M3 plant, we will write the two mutant phenotypes together separated by a slash. For example, dwf/erl means the M3 plant has both dwarf and erect leaf phenotype. If the mutant phenotypes appear in separate plants within a line, we will record them as two sperate entries. The line ID 25M2-0345 is a unique number in the mutant library.

By the way, the line ID for p23 msd1 is 25M2-2176.

The common abbreviation for bloomless mutant is bm, leaf firing is lf, dwarf is dwf, erect leaf is erl.

M2P is the old ID for the M2 plant. Later, I added the EMS concentration to track all mutant lines."

NewcombMaria, 2018-12-04:
@dlebauer and @kimberlyh66 since the S7 RIL experiment is a repeat of S2 experiment (Aug-early Dec 2016), I revised the experiment description for Season2 and used similar wording for the Season7 RIL experiment.  See S2 RIL experiment description https://terraref.ncsa.illinois.edu/bety/experiments/6000000008 

I created a new S7 spreadsheet with cultivar names that match S2, and with revised mutant F2 identities, and with descriptions of four experiments.  I also added an ecotype column as David suggested.  We may need to change the ecotype names if they aren't compatible with other seasons or experiments.  The new spreadsheet is in google drive: 
https://docs.google.com/spreadsheets/d/1RsGLuvewhfeMJbpmB1K3HrYSzwQsbVXI1_GYvk2DxE0/edit#gid=997884579

dlebauer, 2018-12-12:
@NewcombMaria could you provide more details on the "small photosynthesis-inhibitor herbicide experiment to evaluate the PSII imaging system." in the uniformity experiment?

dlebauer, 2018-12-12:
I believe that all of the updates have been made; the sql statements are in the google drive.

Before closing this issue:

* [ ] @kimberlyh66 please check that data are consistent using the R functions that you developed
* [ ] @NewcombMaria I think it would be useful to describe the herbicide experiment in the experimental description ... please update here: https://terraref.ncsa.illinois.edu/bety/experiments/6000000037/edit

NewcombMaria, 2018-12-12:
@dlebauer are you referring to the 'Uniformity Blocks' description?  There were multiple experiments completed in that section of the field, including destructive sampling for in-season biomass in multiple plots, and the validation test for the PSII imaging system.  The uniformity ranges served more than one research objective.  How do you want to handle that?

dlebauer, 2018-12-12:
This is where it would be helpful to understand how the data will be used in order to know how to encode the information. But here are a few options:

- We could subdivide the experiment into specific objectives, and each plot could be linked to one or more experiments. 
- We can treat interventions such as herbicide application and biomass harvests as treatments, associated with a subset of managements. For example, if a subset of plots were harvested early, we could name the treatment 'week 1 biomass harvest' and link those plots to a management that said it was harvested on that date. 
- for the plots used to test the PSII imaging system, unless there were any specific interventions then the fact that the plots were used in this way would be included in the traits table because the any sensor or validation measurement would be associated with the names of the associated plots.

If this is getting too complicated (it is hard for me to know without fully understanding what was done) then lets find a time to chat this week.

NewcombMaria, 2019-08-16:
@dlebauer @kimberlyh66 I was able to set the S7 treatment as "Mac Season 7: Sorghum" and upload most of the field data sets.  One timed out and will need to be subdivided, another needs decisions about how to handle growth stage times that extend beyond harvest date.