terraref/reference-data #251: Hyperspectral calibration target support arm

NewcombMaria, 2018-09-14:
@remotesensinglab an outcome of a productive meeting with Pedro and @jdemieville-ua:  we realized the need for test scans to evaluate calibration targets placed at different positions below the hyperspectral camera lenses.  From an engineering perspective, mounting a target attached to the camera box is most stable if the target can be fully or partially contained within the camera box, which would put the target close to the camera lenses rather than suspended down to the same plane as plant targets.  Jeffrey scripted a series of tests to determine the spectral quality of targets when located closer to the lenses than the plant targets.  The tests should permit better understanding of the sensors and allow us to work towards finding the best mechanism for mounting a target to the camera box for inclusion in all hyperspectral scans.

We completed the test on 2018-09-12.  We need input back from your team to know if there's degredation of the spectral response with decreasing distance between hyperspectral lenses and target.  See table below with the data folder names and measured distance between target and sensor window.  All files can be found in the corresponding directories in the Terraref Globus endpoint (Terraref/ua-mac/raw_data/SWIR  and Terraref/ua-mac/raw_data/VNIR).  We placed the targets on a tripod, at height 115 cm from the ground.  See photos below.  

[image]

[image]
 
Distance 0.352 m
[image]

Distance 2.0 m
[image]

Paheding, 2018-09-14:
Thank you for running this test. We are able to download all the data.  We are doing some analysis now and will share our results with the team.

NewcombMaria, 2019-08-09:
This topic is outdated and can be closed.  If needed a new hyperspectral reflectance panel issue can be opened.

closed 2019-08-09 by NewcombMaria