terraref/reference-data #257: Prioritize processing hyperspectral data from Oct 8 2018

dlebauer, 2018-12-14:
This is a high priority for an analysis that we would like to present in February (so ... this should be the first to go through the pipeline).

From Duke Pauli:

> Dr. Thorp collected spectral data with his ASD FieldSPec pro and a comparison between what the gantry and hand-held would be interesting. The data from those days are particularly interesting because we have accompanying leaf-level compositional data and because we have markers on the leaves so in the images we can get point estimates similar to where the spec readings were manually taken.

dlebauer, 2018-12-17:
@Paheding and @SeanHartling could you please try to run the hyperspectral workflow on the Oct 8 VNIR data from the gantry? @max-zilla has tried and encountered some errors from NCO that he has been unable to debug. The errors that Max got are here: [hyperspectral_errors.txt](https://github.com/terraref/reference-data/files/2687868/hyperspectral_errors.txt), and says "Debug this":

```sh
hyperspectral_workflow.sh: ERROR Failed to translate raw data. Debug this:
ncks -O  --hdr_pad=10000 --no_tmp_fl --trr_wxy=939,1600,22045 --trr typ_in=NC_USHORT --trr typ_out=NC_USHORT --trr ntl_in=bil --trr ntl_out=bsq --trr_in=/home/extractor/sites/ua-mac/raw_data/VNIR/2018-10-08/2018-10-08__13-02-12-387/c053899d-f643-45f9-9fdd-1fa0f6a9bff5_raw /home/extractor/hyperspectral_dummy.nc /tmp/terraref_tmp_trn.nc.pid408.fl00.tmp
```

Paheding, 2018-12-17:
@dlebauer It seems some values are missing since it states, "No value for $TERM and no -T specified
tput", we will look deep into it.

closed 2020-04-04 by dlebauer