terraref/reference-data #258: Upload season 7 leaf chemistry data

dlebauer, 2018-12-17:
Need to upload data from Duke Pauli from Oct 8.

* [ ] put spreadsheets from email on google drive
* [ ] create a worksheet of variables in a worksheet
   * some will already exist, many dont
   * only need 'dry basis'
* [ ] define a [new method](https://terraref.ncsa.illinois.edu/bety/methods/new) that relates to the [NIR feed analyses](https://www.wardlab.com/feed-nir-services.php) from Ward Laboratories
* [ ] use the sampleID to combine result page with plot names 
* [ ] review the [documentation for uploading traits via BETYdb API](https://pecan.gitbook.io/betydbdoc-dataentry/adding-trait-and-yield-data/adding-traits-via-the-v1-api), and prepare a csv to be uploaded
* [ ] add sample date = oct 8 2018

dlebauer, 2019-01-09:
For the existing crude_protein and sugar content variables, we need to distinguish the variable from the method. These data were measured using the John Deer Harvestlab 3000 https://www.deere.com/en/technology-products/precision-ag-technology/data-management/harvest-lab-constituent-sensing/

Prior to uploading the new leaf chemistry data, please:
* [ ] create a new method and citation that describes the HarvestLab and the fact that it is measured on stem + leaf.
* [ ] Update the variable descriptions to remove the reference to the harvestlab method 
* [ ] update the existing trait records of crude_protein and sugar_content to link the harvestlab method

kimberlyh66, 2019-01-11:
@dlebauer thanks for the clarification.  Here is the update statement to link  the existing trait records of crude_protein and Sugar_content to the new harvestlab method. 

```sql
update traits
set method_id = (select id from methods where name = 'Protein leaf and stem measurement')
where variable_id in ((select id from variables where name = 'crude_protein'), (select id from variables where name = 'Sugar_content'));
```

Also, I have temporarily assigned the year to be 2019 in the new harvestlab citation.  Do you know what the year should be?

dlebauer, 2019-01-11:
thanks. If you are referencing a website, use the access date of 2019. If a pdf, either use the publication date or the access date.

dlebauer, 2019-01-11:
I've updated the database.

dlebauer, 2020-06-05:
this has been completed

closed 2020-06-05 by dlebauer