terraref/reference-data #261: Traits package to download data from BETYdb

zhenbinHU, 2019-01-23:
I try to download the date from BETYdb using traits package, for season 1, 2, 5 the code work well. For season 3, I got
>canopy_height_all <- betydb_query(table     = 'search',
                                   trait     = "canopy_height",
                                   sitename  = "~Season 3",
                                   limit     = 'none')
>canopy_height_all
NULL

For season 4, I got
> canopy_height_all <- betydb_query(table     = 'search',
                                   trait     = "canopy_height",
                                   method    = "~Laser 3d",
                                   sitename  = "~Season 4",
                                   limit     = 'none')
Error in betydb_http(url, args, key, user, pwd, ...) : 
  Gateway Timeout (HTTP 504).

dlebauer, 2019-01-23:
There are no canopy height data for season 3. For season 4, the method name is 'Scanner 3d ply data to height' - you should get 63,900 records.

Try 

```r
canopy_height_all <- betydb_query(table = 'search',
                                                            trait = "canopy_height",
                                                            method = "~Scanner 3d",
                                                            sitename = "~Season 4",
                                                            limit = 'none')
```

zhenbinHU, 2019-01-23:
Thanks, it works. Obtained a 70,401 x 38 dataframe.

dlebauer, 2019-01-23:
Great - closing since the issue is solved

zhenbinHU, 2019-01-25:
@dlebauer I used the traits package to download the plant height data. However, how can get a list for traits names for sorghum in betydb. So I can use the traits names to download other trait's data.

dlebauer, 2019-01-25:
You can find a list of all of the variables at terraref.ncsa.illinois.edu/bety/variables or, equivalently, 

```r
betydb_query(table = 'variables')
```

Though note that the existence of a variable does not mean that there are actually data ... 

To see what is actually in the database, you can browse traitvis.workbench.terraref.org and here is a current snapshot of the number of records we have for each variable: https://gist.github.com/dlebauer/5db33c5c54781069ff58302b3313dfd1

closed 2019-02-06 by dlebauer