terraref/reference-data #263: Examination of laser3d height data QA/QC

max-zilla, 2019-02-11:
Wanted to create a main issue to examine the quality of the canopy height trait data derived from the laser3D data.

#262 has some notes from @NewcombMaria about days to examine.

dlebauer, 2019-02-11:
I've marked all of these data in the database as `checked = -1` which means that they aren't available via the api or web interface. These are all of the season 4 and season 6 sample data from ranges 20 and 30, including canopy height and canopy cover: 
[season4and6traits_with_height_and_canopy_cover.zip](https://github.com/terraref/reference-data/files/2853375/season4and6traits_with_height_and_canopy_cover.zip)

max-zilla, 2019-02-28:
Per discussion from email thread:

Roman:
```
Can I get a sanity check on the attached plot? This is showing the
height curves extracted from the current S6 data. Specifically this is
the 95th percentile of the height histograms as a function of day
after planting (there are actually 700 time series plotted here).

My concern is that nothing seems to happen until >30 days in. The
entire field is at a near-constant ~5cm until then.
```

Pless:
```
I think the 5cm arises from a heuristic threshold to quickly distinguish plant from dirt.  Additionally, there seems to be a step function around day 57 for which Zongyang and I are working to track down the reason.
```
Zongyang:
```
In the following picture I am showing point cloud data in the same location that crossing the step day(day 57). The yellow one is before the step(2018-06-14), and the red one is after the step(2018-06-15). It's quite clear to me that top points was cut off on 2018-06-14, then in the next day, gantry goes higher and top leaves show up.
```
[image]

Currently their team is digging further.

max-zilla, 2019-03-14:
it was determined this occurs directly from PLY data, not being introduced from our extractor pipeline.

jdemieville-ua, 2019-04-12:
A series of tests was conducted yesterday with varying scanner height. It should be helpful in determining what happens for plants both higher and lower than intended target height. 

I placed our 3D aluminum target on the third step of a stepladder. Some approximate heights for checking against:
First Step on Ladder: 0.22 m from ground
Second Step on Ladder: 0.48 m from ground
Third Step on Ladder: 0.73 m from ground
Highest surface on 3D aluminum target: 0.83 m from ground
Highest surface on Ladder: 1.15 m from ground

Two series of 11 scans were conducted with target height set to 0.83 m (top of aluminum target). 
In order, the scans were taken as follows:
1. Z = 1.395 (2.5 m from target)
2. Z = 1.595 (2.7 m from target)
3. Z = 1.795 (2.9 m from target)
4. Z = 1.995 (3.1 m from target)
5. Z = 2.195 (3.3  m from target)
6. Z = 2.395 (3.5 m from target; typical height for a target at 0.83 m)
7. Z = 2.795 (3.9 m from target)
8. Z = 3.195 (4.3 m from target)
9. Z = 3.595 (4.7 m from target)
10. Z = 3.995 (5.1 m from target)
11. Z = 4.395 (5.5 m from target)

Associated files for the two series may be found in Globus:
2019-04-11 14:49 - 15:14
2019-04-11 15:16 - 15:40

Looking at the images locally, it seems that when the box is relatively low, then everything above approximately 3 m from the scanners is cut off. There was an old comment in our scripts that "depth of field of the laser is roughly 2.9m until 5.9m distance to the target". When the box is relatively high, it seems like it doesn't fully illuminate the low surfaces. If anyone downstream can make use of this or needs any additional testing to narrow it down, let us know. 

[image]

max-zilla, 2019-04-15:
@ZongyangLi can you take a look at this?