terraref/reference-data #264: Obtaining spectral and composition data at subplot or plant level

ericclei, 2019-02-15:
I am trying to gather data with the BETYdb API at a subplot or plant level. I am most interested in spectral and composition data, though any other categories would be helpful too. The end goal is to predict composition from other traits. As I understand from my conversation with @dlebauer, I will need to process image data to get these fields. Please correct or clarify my statements as needed. I would appreciate any pointers or example scripts to get started.

dlebauer, 2019-02-16:
@ericclei one place to start is with the sample data

- we have sample VNIR hyperspectral and composition data from Season 4 here: https://terraref.ncsa.illinois.edu/clowder/datasets?space=5c50512a4f0c436195b9ad67&status=publicAll 
   - the chemistry data are in this csv https://terraref.ncsa.illinois.edu/clowder/datasets/5c508f7a4f0c65ff4fa3e70d?space=5c50512a4f0c436195b9ad67
   - some hyperspectral data are in https://terraref.ncsa.illinois.edu/clowder/datasets/5c54a8b94f0c65ff4fb6ff02?space=5c50512a4f0c436195b9ad67

I am not sure exactly how to clip the hyperspectral data to a plot, but you can find the plot polygons by constructing a url like: `https://terraref.ncsa.illinois.edu/bety/api/v1/sites?name=~Season 4 Range 30` to find information about all of the season 4 plots in range 30. 

The geometry field is in a standard format called 'WKT' that many different software types can use, it will look something like `"geometry": "MULTIPOLYGON (((-111.97504767486785 33.07488685592014 0.0, -111.97503949029482 33.074886858587995 0.0, -111.97503947576052 33.07485529088314 0.0, -111.97504766033062 33.07485528821534 0.0, -111.97504767486785 33.07488685592014 0.0)))"`

It is possible that there are functions in terrautils for clipping the hyperspectral data but I can not confirm. One challenge is that it has x and y dimensions and the lat and lon information is stored as vectors that map to x and y. If you look inside the netcdf file metadata you can see.

This is not easy but is something that we should be able to add to the terrautils python library if it is not already there. 

At the same time, before using the hyperspectral data you should be aware of the limitations and assumptions in the calibration algorithms, which @Paheding is currently working on improving. See especially https://github.com/terraref/extractors-hyperspectral#limitations as well as links to related issues included at the bottom of that document.

dlebauer, 2019-02-16:
If you use the [nco software](http://nco.sourceforge.net/), you can clip along the x and y dimensions using a command such as `ncks -d x,1200,1400 -d y,2000,2100 in.nc out.nc` to subset a box w/ bounding box (1200,2000),(1400,2100). The challenge is to match up the lat/lon with the y/x dimensions.

This is [documented](https://docs.terraref.org/user-manual/data-products/geospatial-information) but to my knowledge not yet implemented for clipping.

ericclei, 2019-02-19:
Thanks for the guidance. Is there a way to get plant-level chemical composition data, or have I missed something?

dlebauer, 2019-02-20:
@ericclei mostly we are treating the measurements at the plot level. There are a few cases where we have measured traits repeatedly on the same plant. In season 4, some of the plants were tagged for the purposes of repeated measurements. To date, I am not aware that anyone has gone back to identify these specific plants in images although, although in principle I believe it can be done. 

These can be extracted from the database thus:

```sql
copy 
  (select trait, mean, raw_date, entity 
     from traits_and_yields_view 
       where sitename ~ 'Season 4 Range (20|30)'
       and entity != '' and trait not like 'surface_temperature%') 
  to '~/season4_marked_plants.csv' delimiter ',' csv header;
```

I've attached this subsample here and have also placed these in the sample dataset: https://terraref.ncsa.illinois.edu/clowder/datasets/5c6db73e4f0ce7f7828c90a1
[season4_marked_plants.csv.txt](https://github.com/terraref/reference-data/files/2886381/season4_marked_plants.csv.txt)

ericclei, 2019-02-20:
How about composition at the subplot level? How are the plot level measurements collected in the first place?

dlebauer, 2019-02-20:
Information about the methods (still not fully curated) is stored in the [methods table](https://terraref.ncsa.illinois.edu/bety/methods). Often a set of leaves are selected (either randomly, or according to a standard protocol like the three plants at the center of each row) and measured. In general, though, we don't keep track of the individual plants (except to identify that multiple measurements were made on the same plant).

closed 2021-08-18 by dlebauer