terraref/reference-data #266: No Season 8 experiment data in BETYdb

max-zilla, 2019-03-06:
While archiving was going on yesterday I figured I'd start uploading Season 8 raw data to Clowder (we stopped the upload pipeline at the end of Oct 2018 after S7 harvest).

However season and experiment data are returning not found because there doesn't seem to be any S8 definitions in BETY, so I killed the upload until we have some.

Listed experiments:
```
Danforth Sorghum Pilot: Drought Tolerance
MAC Season 1: Biomass
MAC Season 1: Buffer
MAC Season 1: Density
MAC Season 1: RILs
MAC Season 1: Row#
MAC Season 1: Uniform
MAC Season 1: nighthv
KSU 2016: Biomass Association Panel
MAC Season 3: Winter Wheat
KSU 2016: Ashland Bottoms BAP & RIL
MAC Season 1: BAP
MAC Season 3: Winter Wheat Border
MAC Season 3: Winter Wheat Uniformity Block
MAC Season 4: Border Plots
MAC Season 4: Short BAP Accessions
MAC Season 4: Medium BAP Accessions
MAC Season 4: Tall BAP Accessions
MAC Season 3: Winter Wheat North
MAC Season 3: Winter Wheat South
Danforth BAP Drought LT1b: Drought Experiment (TM016_F_052716)
Danforth BAP Drought LT1a:  Drought Experiment (TM015_F_051616)
MAC Season 5: Winter Wheat
MAC Season 6: Sorghum BAP
MAC Season 4: All BAP Accessions
MAC Season 7: Border
MAC Season 7: Stay-green RILs F10
MAC Season 7: Sorghum All Plots
MAC Season 7: Sorghum Mutants, F2 families
MAC Season 7: Pioneer Hybrid Uniformity Blocks
MAC Season 2: Stay-green RILs F10
```

dlebauer, 2019-03-06:
@NewcombMaria can you provide the experimental data and work with @kimberlyh66 to get it uploaded?

NewcombMaria, 2019-03-06:
@kimberlyh66 and @dlebauer I uploaded a spreadsheet with the site/cultivar/treatment metadata to the shared folder: 
https://docs.google.com/spreadsheets/d/1yxf6-ji0Ay4t3GPbqQmYs5znGzys6fdif-NnYNrhs6I/edit#gid=1923709505 

The shape files for the plot polygons have shifted slightly for season 8.  The shape files for the single-row subplots and the two-row plots are in the folder at the following directory: 
https://drive.google.com/drive/folders/15FoffP8bYO_7Ei0rCUER2FQbOjIiKIon 

Let me know if you have questions.

dlebauer, 2019-03-13:
@NewcombMaria what are the planting and harvest dates? Do you have details on the experimental design and irrigation? Are all of the genotypes already in the database?

NewcombMaria, 2019-03-13:
@dlebauer planting date was 2018-12-18, harvest has not happened yet.  All genotypes should be in the database from previous seasons.  The experimental design is a row-column design with two irrigation treatments implemented starting 2018-03-06 (no irrigation and well-watered with irrigation as needed based on weather).

dlebauer, 2019-03-13:
also, for clarification - each two-row plot contains one irrigated and one non-irrigated row? If that is the case, then I think we should exclude the two-row plots since an experimental replicate consists of only a single row. Correct?

NewcombMaria, 2019-03-13:
Yes, starting 2018-03-06 until the end of the field trial a single experimental replicate consists of a single row.  Although from planting until 2018-03-06 both rows of the 2-row plots were under the same irrigation regime and the experimental unit could be considered a genotype (a 2-row plot).