terraref/reference-data #267: How to handle duplicate variable names 'panicle_height' and 'spike_height'

kimberlyh66, 2019-04-12:
These should refer to a single variable?

Please reference terraref/reference-data#118.

dlebauer, 2019-04-16:
panicle_height and spike_height should be combined.

One of the challenges highlighted in #118 is whether these are synonymous with canopy_height.

NewcombMaria, 2019-04-16:
For field data collected in Maricopa 'panicle_height' (ground to top of panicles) is distinct from 'canopy_height' (ground to top of foliar canopy).  

Between the two variable names 'panicle_height' and 'spike_height', my vote for sorghum data is 'panicle_height'.  However I understand if you prefer to go with more generic 'spike_height' which could refer to wheat as well.