terraref/reference-data #268: Radiation interception field measurements

NewcombMaria, 2019-04-12:
Season1 data-set for radiation interception field measurements using a Decagon ceptometer was uploaded to Betydb and subsequently we decided to reformat the variables (incident and ground-level PAR).  We discussed that Season1 data should be removed and then re-entered after converting to fractional PAR. 

Data-sets from other seasons still need to be formatted as fractional PAR and uploaded.

dlebauer, 2019-04-16:
@NewcombMaria could you provide specific instructions about what needs to happen? For example, something like "Delete all records with variable name X, multiply this by that, re-upload". Links to spreadsheets with the data and / or related issues would also be helpful. 

Note that if it is a matter of changing units or multiplying by a constant or another number, it may not be necessary to delete and re-upload.

NewcombMaria, 2019-04-16:
@dlebauer could we have a brief phone or shared-screen meeting on this topic sometime?  It tracks to season-1 data and your suggestion to change the variable to fractional PAR and interpolating incident PAR from sparse time points - it would help to understand your preference for the PAR measurements.

dlebauer, 2019-04-17:
I'm busy today and tomorrow but Friday I have time available. I think it would make sense to store data as ground level PAR with above canopy PAR as a covariate, plus % interception. If we already have the ground and above canopy PAR we can compute the %interception from these without removing data from the database.

dlebauer, 2019-04-26:
Season 1 data are here: https://docs.google.com/spreadsheets/d/1qq7gei6shrFjv91R9Bj92pMp4sIci4ThKZbngSbDOwA/edit#gid=2133044610

David:
* delete existing records from season 1 of incident_par and ground_level_par and Minutes_after_first_Incident_PAR
* add incident_PAR and ground_level_PAR to trait_has_covariates table

Maria:
* interpolate time
* compute the fraction_IPAR_at_0cm
* create a method and document protocol 
* re-upload, with incident_par and ground_level_PAR as covariates
* add measurement_height as a covariate (this is intentionally redundant with name)

dlebauer, 2019-04-26:
Created a new variable, fIPAR https://terraref.ncsa.illinois.edu/bety/variables/6000000339

> fIPAR = 1 - (PAR below canopy) / (PAR at top of canopy) Separate variables include fIPAR_at_25cm = 1 - (PAR at 25cm above ground) / (PAR at top of canopy) Not to be confused with fAPAR, fraction absorbed photosynthetically active radiation with is the 1 - (downwelling PAR)/(upwelling or reflected PAR)

```sql
# delete existing records
delete from traits where variable_id in (select id from variables where name in ('incident_PAR', 'ground_level_PAR', 'Minutes_after_first_Incident_PAR'));
# delete variable that isn't very useful
begin; delete from variables where name = 'Minutes_after_first_Incident_PAR';

insert into trait_covariate_associations (trait_variable_id, covariate_variable_id, required) 
  values ((select id from variables where name = 'fIPAR'), (select id from variables where name = 'ground_level_PAR'), 'f');
insert into trait_covariate_associations (trait_variable_id, covariate_variable_id, required) 
  values ((select id from variables where name = 'fIPAR'), (select id from variables where name = 'measurement_height'), 'f');
insert into trait_covariate_associations (trait_variable_id, covariate_variable_id, required) 
  values ((select id from variables where name = 'fIPAR'), (select id from variables where name = 'incident_PAR'), 'f');
```