terraref/reference-data #269: Need updated Bety data for North field and Season9 plot data

max-zilla, 2019-05-06:


NewcombMaria, 2019-05-17:
@dlebauer the field metadata plot boundaries were checked yesterday by @jdemieville-ua and myself.  The plot boundaries/centroids for Season-9 remained consistent with Seasons 7, 6, and 5.  Season-8 plots differ slightly from previous seasons - S8 shape files are available in a shared folder.  To finalize the S9 site-cultivar spreadsheet I need to find out the standardized ID names that are associated with the genotype and the controlled-environment phenotype data sets for this GRASSL*RIO RIL population.  I've reached out to try to get that info and then I'll finalize the site-cultivar-treatment file.

max-zilla, 2019-05-20:
@nfahlgren @nshakoor should have received an email last Friday regarding this

nfahlgren, 2019-05-20:
Hi @NewcombMaria, sorry for the delay. I looked up the IDs in the controlled-environment experiment and it looks like they used the format that was on the seed packets: `27.5_XXX`.

NewcombMaria, 2019-05-22:
@dlebauer and @max-zilla the Season-9 South field metadata spreadsheet with sites, treatments, and cultivar names (which keep the same cultivar identifier information as the controlled-environment experiment, thanks to @nfahlgren) is uploaded to the shared folder: 
https://drive.google.com/drive/folders/1qqnCCFZWL14S-XhVN5ix0swWdkfUV4RB 

David and I communicated about the S9 site names given South and North fields for the first time.  For S9 the south field ranges are numbered 1-54 and north field ranges are 55-94.  Including 'south field' in the site name along with the range number might be redundant this season.  In future seasons range numbering might change.  

I followed the 'ecotype' categories used in S7. 

Plot boundaries for south field S9 are consistent with the polygons and shape files for Seasons 5, 6, and 7.

dlebauer, 2019-05-23:
@NewcombMaria could you please work with @kimberlyh66 to upload these data?

This would start with making a copy of the [metadata template](https://docs.google.com/spreadsheets/d/1pNEuwNHNHHrpql0b36DYFAbOlBU5TJVF_nO4RdRzdmg/edit#gid=0) that she created and and filling it out. This is also a good chance to provide her with feedback on the structure of the template itself and how easy it is to use, but I think this is a good place to start. If you prefer to add information to the database via the website (e.g. experiments, treatments, etc) then you can skip those sheets in the template. 

@kimberlyh66 we need to be able to re-use plots from prev. seasons where they exist https://github.com/terraref/reference-data/issues/269#issuecomment-493569236, so perhaps make this an option in your function.

NewcombMaria, 2019-05-29:
@kimberlyh66 and @dlebauer I updated the south field metadata spreadsheet to add a description of the experiment and treatments: 
https://docs.google.com/spreadsheets/d/1s7AYUjDLiODpYgURCbjKSji00CLaj_Nw37qly-SrP6I/edit#gid=583364933

https://docs.google.com/spreadsheets/d/1s7AYUjDLiODpYgURCbjKSji00CLaj_Nw37qly-SrP6I/edit#gid=1200493486

All of the information for the metadata template are included.  Let me know if you have questions or need additional info.

NewcombMaria, 2019-05-31:
@dlebauer and @kimberlyh66 the North field and experiment site-cultivar metadata are uploaded to a shared folder.  
The site-cultivar spreadsheet: 
https://docs.google.com/spreadsheets/d/1-taHzWJ1lP_ME4rtJ8SHiia7N27VbZ030uYjoCNvHCE/edit#gid=44530443

Descriptions: 
https://docs.google.com/spreadsheets/d/1-taHzWJ1lP_ME4rtJ8SHiia7N27VbZ030uYjoCNvHCE/edit#gid=1047173486

I think that includes all the information needed for the metadata template.  let me know if you have questions.

NewcombMaria, 2019-05-31:
@dlebauer the **North field plot boundaries** were measured yesterday in gantry coordinates by @jdemieville-ua and myself.  Based on seedling positions after emergence we can confirm that the plot lengths (3.4852 m planted length plus 0.5 m walkway) and widths (0.762 m) are the same as in the South field Season-9 (consistent with South field plot polygons in Seasons 5, 6, and 7).  The row positions along the Y-axis also came out the same as in the South field. 

North field Ranges include ranges 55 through 90 (increasing south to north).  The south end of Range 55 is positioned at X = 229.947 (gantry coordinate system). The north end of the planted area (end of Range 90) is positioned at X = 388.855.

dlebauer, 2019-06-03:
@NewcombMaria do you have the X offset, i.e. how many meters is it from the south end of range 1 to the south end of range 55? Where are the measurements that you and Jeffrey took?

@kimberlyh66 (if you haven't already) please

1. for south field, duplicate season 7 plots and rename them to season 9 for the south field. 
2. For the north field 
   * duplicate season 7 plots
   * add 54 to the range number in the plot names (so they go from 55-94) 
   * use st_translate function similar to https://github.com/terraref/reference-data/issues/194#issuecomment-334316235 to move ranges 55-94 north by the appropriate distance. See documentation for more information https://docs.terraref.org/user-manual/data-products/geospatial-information

NewcombMaria, 2019-06-03:
@dlebauer the following are some offsets to use as anchor points: 

- the south end of Range 54 is at X position 210.02.  The start of Range 55 at X = 229.947 is 19.927 meters north of the start of Range 54. 

- the south end of Range 3 is at X position 6.69.  The start of Range 55 at X = 229.947 is 223.257 meters north of the start of Range 3.

dlebauer, 2019-06-03:
And the south end of Range 1? is that at 6.69 - 2*(3.482 + 0.5) = -1.274?

NewcombMaria, 2019-06-03:
@dlebauer that's correct.  I was using planted length 3.485, so pretty close. South end of Range 1 is at 6.69 - 2*(3.485 + 0.5) = -1.28.

max-zilla, 2019-08-01:
what is the status of this issue?

dlebauer, 2019-08-10:
@NewcombMaria or @jdemieville-ua it is not clear from the discussion above - could you please provide one of the following: 
* the distance from the SE corner of the S field to the SE corner of the N field in meters of offset along N and E?
* a shapefile of these plots (preferred, but likely more work)

NewcombMaria, 2019-08-12:
@dlebauer the offset from the S edge of the Range 1 (S field) to the S edge of Range 55 (N field) is 231.23 m to the north (the distance from Range 3 to Range 55 is 223.257 m, adding 3.985*2 for the additional distance to Range 1).  The rows are parallel to the rails and in line between the S and N fields so there should not be an E/W offset.

NewcombMaria, 2019-08-20:
@dlebauer were you able to create North field plot polygons from the info above?  Can this issue be closed?

dlebauer, 2019-12-16:
Just added:

* experiments_sites relations for full plots in North field
* sites_cultivars relations for full plots in North field
* sites_cultivars for subplots in South field

closed 2020-02-13 by dlebauer