terraref/reference-data #270: Request Input on RGB Camera Settings - Season 9

jdemieville-ua, 2019-05-20:
Title: Request Input on RGB Camera Settings - Season 9

Tags: Stereo, RGB, Exposure, Gain, Gamma, White balance

### Description
We're looking for input on our RGB camera settings. We've previously been told it is better to have the images underexposed than overexposed, but it is unclear to us how dark of an image still retains sufficient data. 

In the time since emergence, we've used a variety of settings in an attempt to minimize overexposure. This has resulted in some images being dark at different sun angles and under cloud cover. I've attached a couple measurements with the corresponding settings files. One (2019-05-17__09-39-47-766) is with a setting used in prior seasons (f6,7_summer_shade) and one (2019-05-19__09-12-05-240) is with a newer setting that's been tweaked (StVis_exp1450_gain1150_gamma60). For much of the time from 09:30-13:00, the area under the camerabox is shaded. An unshaded area starts to develop as the afternoon progresses.

[image]

I'm looking for confirmation that useful data can still be extracted from these darker measurements and that it isn't being lost due to underexposure.

[Attached data containing raw measurements, settings files for reference, metadata, and locally-produced preview images](https://drive.google.com/open?id=1FuLd2udiNd5_yw_BlyC-3RQmlk6juAtW
)

Additionally, we've found that some of the settings files (e.g. f6,7_summer_shade) are using different white balance ratios for each camera. Would it be preferable for the pipeline to attempt to adjust these white balance ratios for visual consistency between cameras, or for a single set of ratios to be selected and kept constant?

@ZongyangLi , @pless , @abby621 , @NewcombMaria

jdemieville-ua, 2019-06-20:
Follow-up to email conversations requesting images to be partly shaded and partly sunlit

[image]

Appears clear that f6,7_shade results in overexposure on leaves in sunlit portion. Leaves still look a little overexposed on the others, but we're curious if the shaded area would still be useful if the exposure is low enough to not overexpose.

ZongyangLi, 2019-06-24:
@jdemieville-ua 

When I test my segmentation algorithm, the right side images works well for precious criteria, even in the shaded area. It would still be useful in this case.

closed 2019-07-29 by jdemieville-ua