terraref/reference-data #273: BETYdb S6 manual leaf length/width units

NewcombMaria, 2019-08-09:
@dlebauer and @kimberlyh66 when I uploaded the manual field measurements to Betydb for Season 6 leaf length and width the values had units of cm instead of mm.  Abby pointed out the difference between S4 data which are mm and S6 data that need to be converted from cm to mm. The data file is at the following directory: 
https://docs.google.com/spreadsheets/d/1JahNn4wmhHIjET2SuiC0g-_1ujgxrS7W3YD5VEgltXk/edit#gid=2088778731

Is it possible to make the conversion in Betydb directly?

dlebauer, 2019-08-19:
need to update season 4 data 

```sql
begin;
update  traits  set mean = mean * 10 
WHERE
	variable_id IN (
	SELECT ID 
	FROM
		variables 
	WHERE
	NAME IN ( 'leaf_width', 'leaf_length' ))
	AND citation_id IN ( SELECT ID FROM citations WHERE author LIKE 'Newcomb%') 
	and site_id in (select id from sites where sitename like '%Season 4%');
commit;
```

dlebauer, 2019-08-19:
these have been updated

NewcombMaria, 2019-08-20:
@dlebauer from your code it looks like values for Season 4 were changed.  Season 6 leaf length and width trait values are the ones that need to be converted from cm to mm.  The Season 4 leaf length/width were originally in the correct units.

dlebauer, 2019-08-20:
thanks for catching that - I've just applied this fix

### undo above

```sql
begin;
update  traits  set mean = mean / 10 
WHERE
	variable_id IN (
	SELECT ID 
	FROM
		variables 
	WHERE
	NAME IN ( 'leaf_width', 'leaf_length' ))
	AND citation_id IN ( SELECT ID FROM citations WHERE author LIKE 'Newcomb%') 
	and site_id in (select id from sites where sitename like '%Season 4%');
commit;
```

### Apply cm --> mm to Season 6

```sql
begin;
update  traits  set mean = mean / 10 
WHERE
	variable_id IN (
	SELECT ID 
	FROM
		variables 
	WHERE
	NAME IN ( 'leaf_width', 'leaf_length' ))
	AND citation_id IN ( SELECT ID FROM citations WHERE author LIKE 'Newcomb%') 
	and site_id in (select id from sites where sitename like '%Season 6%');
commit;
```

closed 2019-08-20 by dlebauer