terraref/reference-data #276: BETYdb upload field measurements S9 South field

NewcombMaria, 2019-09-10:
@dlebauer and @Kimberly Huynh a couple 2019 S9 South field data sets are ready to be uploaded to Betydb.  I tried the bulk upload process and got an error about unresolvable site and treatment references.  Have culivars been linked to sites for S9 south field?  https://docs.google.com/spreadsheets/d/1s7AYUjDLiODpYgURCbjKSji00CLaj_Nw37qly-SrP6I/edit#gid=1200493486

kimberlyh66, 2019-09-11:
@NewcombMaria cultivars have not been linked to season 9 sites. I will work on getting those linked and let you know when I finish.

NewcombMaria, 2019-09-27:
@kimberlyh66  and @dlebauer are there still some steps that need to be taken for S9 field data uploads?  I tried again to upload the data set for South field final biomass but still get errors for unresolvable site references and treatment references.  

[image]

dlebauer, 2019-09-27:
what are the warnings?

NewcombMaria, 2019-09-27:
The warnings are only a list of columns that will be skipped, which is expected.

NewcombMaria, 2019-10-10:
Hi @dlebauer and @kimberlyh66.  Any updates on this issue?  I'd like to follow through on the south field S9 upload for the 2 data sets either today or tomorrow if possible.

dlebauer, 2019-10-10:
@NewcombMaria where is the data that you are trying to upload?

NewcombMaria, 2019-10-10:
https://drive.google.com/drive/folders/1qqnCCFZWL14S-XhVN5ix0swWdkfUV4RB 
2 data sets as csv files

dlebauer, 2019-10-10:
It looks like in the metadata Kimberly uploaded, plots were divided into east and west columns; in the data you are trying to upload it is not divided by E/W. Is that correct?

NewcombMaria, 2019-10-10:
That's correct.  Both data sets are plot-level values (2-row plots).

dlebauer, 2019-10-10:
@NewcombMaria I've fixed a few things - if you remove the string ' South Field' from your plot names, it should work.

NewcombMaria, 2019-10-10:
Success!  I had to add a treatment called 'Border' for the buffer plots in between the 2 different irrigation levels.  Empty cells in the treatment column were not accepted.  The Season-9 south field biomass harvest data are uploaded along with a small subset of manual plant height estimates.

dlebauer, 2019-10-10:
Fantastic! Can this issue be closed?

NewcombMaria, 2019-10-10:
Yup, thanks

closed 2019-10-10 by NewcombMaria