terraref/reference-data #278: Generate Season 10 (Lettuce) Plot Polygons

jdemieville-ua, 2019-12-16:
This issue is created to capture discussion in Slack regarding plot polygons for season 10 Lettuce. Start and end positions, average length, and initial start position for ranges determined in gantry coordinate system. Row positions also collected. Work performed per standard protocol [(link)](https://drive.google.com/open?id=1NZr5kq2WaW9Pq-gWsXwe3Ch6SEi04qxRABjuHV_D-gw). Data compiled in Google Spreadsheet at [this location](https://drive.google.com/open?id=1yTTIrbX-SZyadNl5yxB5jkGLUANZbfVq7KNpa9WDVR8). Refer to the second page of the document (fieldPositionsS10) for information relevant to pipeline. 

As a comment from Maria Newcomb, "Andy French deserves much of the credit for previous plot polygon shape files, pretty much a team effort with Rick sharing drone mosaics, Max providing gantry rgb full field stitches, Jeffrey and Pedro helping collect gantry coordinate anchor points and planted area corner points". 

@dlebauer was following up with Andy through email. 

 
### Completion Criteria 
* [ ] Shapefiles generated and integrated into pipeline

dlebauer, 2020-01-14:
all of the metadata are here and ready for upload to BETYdb

rleontie, 2020-02-27:
The metadata seems wrong: see examples for "created_at": "2017-07-11 16:37:19 -0500",
and the "sitename": "MAC Field Scanner Season 4 Range 51 Column 13 E", that reference season 4. 
[image]

dlebauer, 2020-02-27:
Sorry - I gave you the wrong url and wrong api call since we are using a new database for the Open project

closed 2020-02-27 by dlebauer