terraref/reference-data #281: upload leaf width / height for seasons 4 and 6

dlebauer, 2020-02-06:
We would like to update the results on Betty DB for the leaf length and width extractors for season 4 and 6 data. You can find [season 4](https://drive.google.com/file/d/1ud_gnJTEIGAd10a7bseTn8x9oBaFHbrv/view?usp=sharing) and [season 6](https://drive.google.com/file/d/1JP5JpgnlhoBe3iG53Q5BeA8mBFKsi1ui/view?usp=sharing) linked here.

rleontie, 2020-02-06:
Confirming please delete existing records. The measurements are in millimeters so not sure that having more than 3 digits make sense.

dlebauer, 2020-02-06:
## Flagging old records

```sql
update traits set method_id = 6000000040 where method_id = 6000000029;
delete from methods where id = 6000000029;

update traits set checked = -1, 
notes = 'old version of leaf length / width extractor replaced 2020-02-14', access_level = 1
WHERE
	site_id IN ( SELECT ID FROM sites WHERE sitename LIKE '%Season 4%' ) 
	AND variable_id IN ( SELECT ID FROM variables WHERE name IN ( 'leaf_width', 'leaf_length' ) ) 
	AND method_id IN ( SELECT ID FROM methods WHERE name = '3D scanner to leaf length and width' );

update traits set checked = -1, 
notes = 'old version of leaf length / width extractor replaced 2020-02-14', access_level = 1
WHERE
	site_id IN ( SELECT ID FROM sites WHERE sitename LIKE '%Season 6%' ) 
	AND variable_id IN ( SELECT ID FROM variables WHERE name IN ( 'leaf_width', 'leaf_length' ) ) 
	AND method_id IN ( SELECT ID FROM methods where name = '3D scanner to leaf length and width' )
```

## Uploading new data

```sh
curl -X POST --data-binary @betydb_s4_leaf_length_width.csv https://terraref.ncsa.illinois.edu/bety/api/v1/traits.json?key=SECRET
curl -X POST --data-binary @betydb_s6_leaf_length_width.csv https://terraref.ncsa.illinois.edu/bety/api/v1/traits.json?key=SECRET
```

dlebauer, 2020-02-06:
try dividing into chunks ...

```sh
sed -n -e 1p -e 2,5000p s4.csv > s4.1.csv
sed -n -e 1p -e 5001,10000p s4.csv > s4.2.csv
sed -n -e 1p -e 10001,12759p s4.csv > s4.3.csv
```

```sh
for i in {1..3}; do echo curl -H "Content-Type: text/csv" -X POST --data-binary @s4.$i.csv https://terraref.ncsa.illinois.edu/bety/api/v1/traits.csv?key=EQ7Gk6ZB7CT2LXdAPhGW2xuojxfbRDxTrKqdVGJI; sleep 10m; done;
```

closed 2020-02-07 by dlebauer