terraref/reference-data #282: upload emergence count for seasons 4 and 6

rleontie, 2020-02-06:
@dlebauer please assist in adding the results for season 4 and 6 emergence count on Betty DB.   Data can be found [here](https://drive.google.com/open?id=1Tz9UsjTNMcN9-igNr1FVDHg-zoJ5MpCg)

dlebauer, 2020-02-06:
```sh
# fix trait name 
for f in *.csv; do sed 's/counting/count/g' $f > new$f; done;

# upload
for f in new*.csv; do curl -H "Content-Type: text/csv" -X POST --data-binary @$f https://terraref.ncsa.illinois.edu/bety/api/v1/traits.csv?key=EQ7Gk6ZB7CT2LXdAPhGW2xuojxfbRDxTrKqdVGJI; done;

```

@ZongyangLi please update the [citation](https://terraref.ncsa.illinois.edu/bety/citations/6000000026) and [method](https://terraref.ncsa.illinois.edu/bety/methods/6000000089):

* [ ] Citation should include a link to the github repo or zenhub doi
* [ ] method should describe how this is done

dlebauer, 2020-02-07:
@rleontie should I also delete existing data from this?

rleontie, 2020-02-07:
@dlebauer I don't know. @ZongyangLi do we need to keep the existing data for this?

ZongyangLi, 2020-02-07:
@dlebauer I am not clear about the requirement of updating `citation` and `method` you mentioned above.
For citation, we have `citation_author`, `citation_year`, `citation_title` already. 

> it is necessary to have either a citation_doi column, or have all three of the columns citation_author, citation_year, and citation_title.

For method, the current description is `Stereo RGB data to emergence counting`, would you like a much longer description to describe the whole thread?

About the existing data, do you mean the data we just uploaded yesterday?

dlebauer, 2020-02-07:
@ZongyangLi 

- for citation, you've listed the minimum requirements required by the database, but for scientists interested in how the data was generated, it would be more helpful if there is a link to the github repository where the code is, preferably with a version that can be archived on Zenodo. If you can point me to the version that was used, I can do the rest.
- for the description, it would be helpful to have a brief summary of the method used in the database, and then the citation field could link to the repository with a more technical description in the README similar to what you have provided in the past.

- for existing data - was there any emergence data previously uploaded that this is replacing?

ZongyangLi, 2020-02-07:
@dlebauer 

My code uploaded to : https://github.com/terraref/extractors-stereo-rgb/tree/stereo-emergence-detection/emergence_detection

I will write a more technical description later and let you know.

There was not any emergence data previously. This is the first time we create these csvs.

dlebauer, 2020-03-07:
@ZongyangLi can this be merged? https://github.com/terraref/extractors-stereo-rgb/pull/55 I replaced "counting" with "count" so that it would generate the correct variable and methods name (above I had to find/replace to get the files to upload)

dlebauer, 2020-04-04:
data have been uploaded.

closed 2020-04-04 by dlebauer