terraref/reference-data #283: canopy cover season 4

rleontie, 2020-03-06:
@dlebauer please upload canopy cover extractor results to BettyDB. 

Results generated by @ZongyangLi can be found [here]( https://drive.google.com/open?id=1DCs4f3xw-xiMk0_HS1elJOe2ZYebV3kk)

dlebauer, 2020-03-06:
should I upload and replace existing data?

dlebauer, 2020-03-06:
On its way. I've converted from fraction [0,1] to percent [0,100] and truncated values to 3 significant figures:

```r
filenames <- dir("~/Downloads/canopy_cover_season4", full.names = TRUE)
require(data.table) ## 1.9.2 or 1.9.3
data <- rbindlist(lapply(filenames, fread))
hist(data$canopy_cover)

canopy_cover <- data %>% mutate(canopy_cover = signif(canopy_cover*100, digits = 3))
readr::write_csv(canopy_cover, '~/Downloads/s4canopy_cover.csv')
```

command used to upload data:

```sh
curl -H "Content-Type: text/csv" -X POST --data-binary \
  @s4canopy_cover.csv https://terraref.ncsa.illinois.edu/bety/api/v1/traits.csv?key=
```

dlebauer, 2020-03-07:
Uploading finished - 54710 new records!

closed 2020-03-07 by dlebauer

dlebauer, 2020-03-07:
updating treatments

```
begin;
update traits set treatment_id = 6000000024 where site_id in (
select id from sites where sitename ~ 'Column (5|6|7|12|13|14|15|16)$' 
              and sitename like 'MAC Field Scanner Season 4%') and treatment_id is NULL;
update traits set treatment_id = 6000000026 where site_id in (
select id from sites where sitename ~ 'Column (1|2|3|4|9|10|11)$' 
              and sitename like 'MAC Field Scanner Season 4%') and treatment_id is NULL;
commit;
```