terraref/reference-data #285: How to properly convert lemntec rawx format to generic hdr?

parth-sankhavara, 2020-04-15:
I have rawx image files of lemnatec phenotyping. I need to convert back into hdr format. How could I?

dlebauer, 2020-04-15:
@parth-sankhavara I don't think we have worked on that - we have hyperspectral cameras that generate hdr files. Where did you get these files?

parth-sankhavara, 2020-04-15:
> @parth-sankhavara I don't think we have worked on that - we have hyperspectral cameras that generate hdr files. Where did you get these files?

@dlebauer No Problems, Btw, I have got that rawx files from lematec software.

dlebauer, 2020-04-15:
got it - even though we use lemnatec hardware, they didn't provide any software for data processing - so we mostly start with whatever files the sensors generate and process from there, which is hdr in the case of the headwall sensors.

closed 2020-04-15 by dlebauer